jcurl/RJCP.DLL.SerialPortStream #1: Parity setting

robert-willis, 2016-03-14:
Using the default constructor, if i set parity before opening,(in my case to odd), once i open the port the parity becomes none, and I have to  set the parity to odd after opening the port for it to work.

using the constructor SerialPortStream(string port, int baud, int data, Parity parity, StopBits stopbits) also fails in keeping the parity setting once the port is open.

jcurl, 2016-03-14:
Hi,

Can you please provide details about the chipset you use? Are you using master or a particular version?

robert-willis, 2016-03-15:
Hi

I’m using a PL-2303 using the most recent driver.
But since you asked, I have since tried com0com serial port emulator, a moxa com port, and ftdi, and the error happens on each.

I am using the dll compiled August 1st 2015, version 1.1.4.0, but I also compiled both the master and the 1.x and tried those too with the same result.

       using (SerialPortStream porto = new SerialPortStream("COM12", 4800, 7, Parity.Odd, StopBits.One))
       {
              porto.Parity = Parity.Odd;
              Console.WriteLine("Before port open, parity = " + porto.Parity);
              porto.Open();
              Console.WriteLine("After port open, parity = " + porto.Parity);
              porto.Parity = Parity.Odd;
              Console.WriteLine("After re setting parity to odd, parity = " + porto.Parity);
              porto.Close();
              Console.WriteLine("After port close, parity = " + porto.Parity);
              porto.Open();
              Console.WriteLine("After port open, parity = " + porto.Parity);
              porto.Close();       }
My results:

Before port open, parity = Odd
After port open, parity = None
After re setting parity to odd, parity = Odd
After port close, parity = Odd
After port open, parity = None

So opening the port resets parity to what seems a default of none.

Robert Willis
Computer and Control Systems Manager
Department of Chemical Engineering
Louisiana State University
office 225-578-3051

jcurl, 2016-03-15:
I've confirmed the bug with a simple test case. I need to look in more detail, initial analysis says that I'm setting the DCB with SetCommState, and then I call GetCommState, but the bit fParity is cleared which is why the software thinks there's no parity.

It could take a few days for me to implement a proper test case over a NULL modem to identify and resolve the issue.

robert-willis, 2016-03-15:
OK , no worries.

For now I can get around by re setting it after open.

Let me know how it goes, or if you want me to test something.

Thanks,

Robert Willis
Computer and Control Systems Manager
Department of Chemical Engineering
Louisiana State University
office 225-578-3051

jcurl, 2016-03-15:
Further, I've confirmed that the only workaround is to set it after opening. I don't know why this bit isn't being properly set from the DCB. I've confirmed by a loop back setting the transmitter to 7,O,1, receiver to 8,N,1 and checking what's received. I'll reply with a comment and close this when I'm finished. The fix will be on the v1.x branch.

robert-willis, 2016-03-15:
Thanks for the update.

Robert Willis
Computer and Control Systems Manager
Department of Chemical Engineering
Louisiana State University
office 225-578-3051

jcurl, 2016-03-15:
So GitHub automatically closes this topic. Interesting, that I didn't have a chance to ask you to test against v1.x branch.

closed 2016-03-15 by jcurl

robert-willis, 2016-03-15:
Yes, it is working. THanks for the update!