jcurl/RJCP.DLL.SerialPortStream #2: "Unable to clear the serial break state" exception in Open()

jwoodwardtfx, 2016-04-05:
When calling SerialPortStream.cs:Open() on a serial port whose driver does not support clearing the serial break state an exception is generated and the port cannot be opened.

There is an attempt to catch this exception when setting the "BreakState" in SerialPortStream.cs:Open() but the exception is actually first generated before this point by the call to "m_NativeSerial.SetPortSettings()" (towards the end of this function the break state is set or cleared, causing the exception to be generated).

```
[SerialPortStream.cs]
        private void Open(bool setCommState)
        {
            if (IsDisposed) throw new ObjectDisposedException("SerialPortStream");
            if (IsOpen) throw new InvalidOperationException("Serial Port already opened");

            m_NativeSerial.Open();
            try {
                if (setCommState) {
                    m_NativeSerial.SetPortSettings(); **<--- EXCEPTION HERE**
                    // Fetch the actual settings and get the capabilities
                    m_NativeSerial.GetPortSettings();

                    try {
                        m_NativeSerial.BreakState = false;
                    } catch (System.IO.IOException) {
                        // Ignore IOException. Not all serial port drivers support clearing the
                        // Break signal, so we ignore it when opening.
                    }
                }
```

```
[WinNativeSerial.cs]

        /// <summary>
        /// Writes the settings of the serial port as set in this object.
        /// </summary>
        /// <exception cref="ObjectDisposedException"/>
        /// <exception cref="InvalidOperationException">Port not open.</exception>
        [System.Diagnostics.CodeAnalysis.SuppressMessage("Microsoft.Usage", "CA2201:DoNotRaiseReservedExceptionTypes")]
        public void SetPortSettings()
        {
            if (m_IsDisposed) throw new ObjectDisposedException("WinNativeSerial");
            if (!IsOpen) throw new InvalidOperationException("Port not open");

...snip...

            if (m_BreakState) {
                m_CommModemStatus.SetCommBreak();
            } else {
                m_CommModemStatus.ClearCommBreak(); **<--- EXCEPTION FROM HERE**
            }
        }

```

jcurl, 2016-04-05:
Thanks for the bug report. Clearly a regression from 1.x. Can you let me know what chipset you use and if it's easily available for my own testing?

jwoodwardtfx, 2016-04-05:
It's a USB cellular modem (u-blox TOBY L210) that enumerates as a number of COM ports.

Probably not easy to get hold of for you to test, but if you have any changes you want me to try then just let me know...

networkfusion, 2016-04-05:
That sounds a lot like the issue I had back along... Try opening direct

jcurl, 2016-04-05:
Can you try applying this patch if this fixes your problem? I don't have my equipment with me to test.

[0001-Fix-exception-on-Windows-for-serial-drivers-that-don.patch.txt](https://github.com/jcurl/SerialPortStream/files/204415/0001-Fix-exception-on-Windows-for-serial-drivers-that-don.patch.txt)

jwoodwardtfx, 2016-04-05:
Yep, that fixes it (it's the same as the fix I've made locally :p).

Thanks.

closed 2016-04-05 by jcurl