jcurl/RJCP.DLL.SerialPortStream #5: lost SerialDataReceivedEventHandler

nidheg666, 2016-04-25:
the transition from version 1.2 from nuget to 2.x from github, lost SerialDataReceivedEventHandler.

Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0246	The type or namespace name 'SerialDataReceivedEventHandler' could not be found (are you missing a using directive or an assembly reference?)	SerialPortCommunication	C:\work\C#\SerialPortCommunication\MyComm.cs	300	Active

using RJCP.IO;
using RJCP;
using RJCP.IO.Ports;

jcurl, 2016-04-25:
There is a slight change in the naming of the events to modernise.

Look at this test case:
        [Test]
        [Category("SerialPortStream")]
        public void WaitForRxCharEventOn1Byte()
        {
            using (ManualResetEvent rxChar = new ManualResetEvent(false))
            using (SerialPortStream src = new SerialPortStream(c_SourcePort, 115200, 8, Parity.None, StopBits.One))
            using (SerialPortStream dst = new SerialPortStream(c_DestPort, 115200, 8, Parity.None, StopBits.One)) {
                src.Open(); src.WriteTimeout = c_TimeOut; src.ReadTimeout = c_TimeOut;
                dst.Open(); dst.WriteTimeout = c_TimeOut; dst.ReadTimeout = c_TimeOut;

                dst.ReceivedBytesThreshold = 1;
                dst.DataReceived += (s, a) => { rxChar.Set(); };
                src.WriteByte(0x00);
                Assert.That(rxChar.WaitOne(1000), Is.True);
            }
        }

The type has changed slightly to use Generics:
        public event EventHandler<SerialDataReceivedEventArgs> DataReceived;

nidheg666, 2016-04-25:
thanks. It came alive)

closed 2016-04-25 by nidheg666