jcurl/RJCP.DLL.SerialPortStream #6: WinNativeSerial.IsOpen flag is always true

vadrianc, 2016-04-25:
After physically severing the serial connection, the WinNativeSerial.IsOpen property is always true. 
I have to mention that I'm using a USB to Serial adapter for testing. Regardless if I remove just the serial cable or disconnect the USB to Serial hub from the PC, the IsOpen flag is always true.

Regards,
Adrian V.

jcurl, 2016-04-27:
I have to check, but when removing a device, I would guess that the main loop in CommOverlappedIO would die. There's an internal flag called IsRunning that probably should be checked.

However, the IsOpen flag should (and must) remain true, even if the device is no longer present, as it's the responsibility of the class calling Open() to also call Close() or Dispose(). The SerialPortStream should not close itself. A similar example can be seen in Network sockets. The remote side closes and all operations should result in an exception, but doesn't close itself.

It should however detect such a situation and raise exceptions (probably IOException) on Read/Write/Flush and other methods that might indicate that the connection is lost.

jcurl, 2016-04-28:
So they way Read() will work is that if the device is returned, it will not raise an exception immediately, but return the number of bytes in the buffer, or zero for end of file. This means if you use blocking behaviour, you can conveniently know when we got to the end by reading zero. That's the specification and behaviour of other streams.

To not confuse a return value of zero on Read() with that of a timeout, the first return is zero, the second is an IOException which is similar to how sockets work (they return zero on read, the second is EPIPE).

Write should raise an exception immediately.

Now to write and test :)

jcurl, 2016-05-10:
Please test. Closing

closed 2016-05-10 by jcurl