jcurl/RJCP.DLL.SerialPortStream #9: Write large data(~128kB) will cause out buffer wrap around

Kurapikov, 2016-06-30:
Hi Jason,

I tested the 2.0.0 release, and found a weird bug.

I need to write about 1MB data through UART. Each time I write 256 bytes (as a page) and wait for slave ack. When about 128kB data is writed, slave received the first page I've wrote.

I tried to flush and discard out buffer after every 256 bytes write, and this problem is resolved.

I have no time to read through the code to analysis what happened.

I can do more test if you need.

Regards.

jcurl, 2016-07-01:
Can you please provide a small test program that recreates the issue? I'm particularly interested in the drivers and chipset of the serial port, how you initialise the serial port. Then please enable debugging to a file and send me the logs (see http://serialportstream.codeplex.com/wikipage?title=TraceSwitch%20Settings&referringTitle=Documentation). This page might also help (http://serialportstream.codeplex.com/wikipage?title=Providing%20Debug%20Information&referringTitle=Documentation)

jcurl, 2016-08-05:
As I've not heard any response, and I've not seen this issue in the unit test cases that do similar work, I'm closing.

closed 2016-08-05 by jcurl