jcurl/RJCP.DLL.SerialPortStream #12: Long delay when opening a serial port (around 30 seconds)

tbayne, 2016-12-27:
I am seeing a long delay when doing a .Open() on a serial port stream.
Delays of 15 to 30+ seconds are common.

A run of the VS 2015 performance profile shows that the delay appears to be in:

Ports.Native.Windows.CommModemStatus.ClearCommBreak().

Is this expected?  
Note that this a USB virtual COM port.

jcurl, 2016-12-27:
Hello Terry,

I've not observed this behavior myself. Can you describe a little more of your setup? What driver are you using and what is the actual hardware? Have tried different drivers, like FTDI or Prolific?

I expected the method to just internally clear flags and not be more than the order of milliseconds. Do you use hardware flow control at all?

tbayne, 2016-12-28:
Jason,

Thank you for the quick response.

The device I am using is a "FlexControl" from FlexRadio Systems.  It came with a a VSP driver, which is what I am using.  (Running at 9600, 8,None,1) on COM99.  No flow control is turned on.  Note that once the port is opened it works perfectly.

This is something I can live with (and potential users of my utility aren't likely going to want to change drivers).  I just wanted to check that this wasn't expected behavior.  

I will put this on the back burner for now and may try to get the device working with alternate drivers in the future.  Once I do, I will report back my findings.

Happy New Year
Terry

jcurl, 2016-12-29:
Hello, I'm attaching a test binary. I've removed the code that calls this method on open and would like your confirmation that it works for you as expected.

[SerialPortStream-2.0.3-Debug-0a840dc.zip](https://github.com/jcurl/SerialPortStream/files/677131/SerialPortStream-2.0.3-Debug-0a840dc.zip)

tbayne, 2016-12-29:
Jason,

I will try this out this afternoon and report back.

Terry

tbayne, 2016-12-29:
Jason,

I managed to fit this in this morning.  I can report that it works much better - the Open() call now takes on the order of 19ms rather than 30000ms - much improved.

I call this a win!

Terry

closed 2016-12-29 by jcurl