jcurl/RJCP.DLL.SerialPortStream #23: Documentation missing

MrEngineer02, 2017-05-02:
I had trouble figuring out how to use this library until I found the documentation on the CodePlex project:
[https://serialportstream.codeplex.com/documentation](url)

I see most of those examples have not been copied to the Wiki on this project:
[https://github.com/jcurl/SerialPortStream/wiki](url)

 Is there any way that I could edit the Wiki to copy/paste the documentation from CodePlex?

-David

jcurl, 2017-05-02:
Duplicate of https://github.com/jcurl/SerialPortStream/issues/16

MrEngineer02, 2017-05-03:
Completed copying documentation from CodePlex.  Thanks for letting me contribute!  My apologies for any disorder.  I'm glad I got to "rescue" all the value-able documentation.

You can close this issue.  I don't expect I will need edit access to the wiki any more.

jcurl, 2017-05-03:
Thank you very much for your contribution.

closed 2017-05-03 by jcurl