jcurl/RJCP.DLL.SerialPortStream #26: DiscardInBuffer

MatejZem, 2017-05-05:
Hello,
First of all thumbs up for the good work.
Second I got an 
```
Unhandled exception System.NotImplementedException in mscorlib
System.NotImplementedException
The method or operation is not implemented.
```
when using the method DiscardInBuffer with Mono on Linux.
This method worked fine before. Is maybe the issue in the new update?

jcurl, 2017-05-06:
Looks like it works in Windows, but was never implemented for Linux. I'll investigate.

jcurl, 2017-05-07:
I've provided a fix if you can test on the branch "bugfix/discard". When you confirm it works as expected, please let me know and I'll merge to v2.x

MatejZem, 2017-05-07:
Thank you for your quick response.
But how do I test it out with my project in Visual Studio? I don't have any dll to reference to.
I had build the .so files for Linux.

Sorry for my lack of knowledge.

jcurl, 2017-05-08:
Hi, no worries. There were no changes to libnserial.so, only to the sources. If you need, I can build for you a test nuget package.

MatejZem, 2017-05-08:
Please build a test nuget package. I will appreciate it alot.

jcurl, 2017-05-08:
The NUGET packages are installed. I usually push the version with symbols to NuGet, so rename that file and put in your path that NuGet can upgrade to within the MonoIDE.

[serialportstream.2.1.2-rc-1-discard.zip](https://github.com/jcurl/SerialPortStream/files/984497/serialportstream.2.1.2-rc-1-discard.zip)

MatejZem, 2017-05-09:
Thank you very much! Is working! 
I had installed package with symbols and for now it is working. Buffer is emptied.
I'm continuously working with Mono on Linux and your Serial Port communication and I will report more bugs, if I may found some.

jcurl, 2017-05-09:
Please do report any issues you find. I'm closing this as you tested the commit id ba916671016c92ad1e143257fb7b98df755c2180 which is now in the v2.x branch.

closed 2017-05-09 by jcurl