jcurl/RJCP.DLL.SerialPortStream #27: What are these dependecies ?

ttustonic, 2017-05-06:
Hello

I'm trying to update/install SerialPortStream 2.1.1, but there is a large list of dependencies which were not there in the version 2.0.3 (see attached image). Why are these and how to get rid of them? Is that a problem in the NuGet package, because I can't see these dependencies in the source.
I'm using Visual Studio 2015 and .Net 4.6.2 project, but it makes no difference if I choose 4.5.2.
[image]
.

jcurl, 2017-05-06:
Those dependencies are only required for .net standard 1.5. There should habe been rules in the nuget package that pull these in only for .net standard. Can you look in your project to be sure you don't use dot net standard? Check the packages.config file, it should say target="net45"

jcurl, 2017-05-06:
I must also say that I have not observed this with my .NET 4.0 projects.

jcurl, 2017-05-06:
I've just created a new project based on .NET 4.6.1. I then installed in this empty project SerialPortStream 2.1.1 from NuGet. It does not install the dependencies for this fresh project.

Further, on that empty project, if I install v2.0.3, then upgrade to 2.1.1, I still don't get those dependencies.

So for some reason, I suspect your project is pulling in .NET Standard and the VS IDE is trying to satisfy these dependencies.

Closing as not reproducible

ttustonic, 2017-05-06:
Hm, I tried the same with Visual Studio 2015 and different .Net versions from 4 to 4.6.2. It seems that this behavior show only in 4.6.2 project. TargetFramework says "net462" in project with 4.6.2, "net461" with 4.6.1. etc. I could probably downgrade my projects to 4.6.1 or earlier, so it's probably not a big deal.

jcurl, 2017-05-06:
Ok, only tried up until 4.6.1 as I haven't installed the sdk for later. Still, I created a new console project for 461 and it worked flawlessly. Project for net40 can't use net standard 1.5. I tested with vs2015 enterprise.

closed 2017-07-15 by jcurl