jcurl/RJCP.DLL.SerialPortStream #29: Linux Arm

nicolasr75, 2017-06-20:
I am so happy to have found this library!
I would like to use it with .NET Core on a Raspberry PI.
So far I was able to compile it with .NET Core on my Windows machine.
As I am not (yet) a Linux expert:
Your libnserial debian packages only include x86, x64.
Do you think there will be any problems compiling it on an ARM86 (Raspberry PI) with Ubuntu 16.04?

jcurl, 2017-06-21:
Have a look in the Dll directory. It uses Cmake for the build system. I have to be honest and say I haven't investigated cross compiling for Cmake, you'll have to research the forums.

You can also get GoogleTest and run the unit tests on the rpi with a loopback cable to see how well it works. Unfortunately I don't have a target to cross compile to so that I can test. Bit if you find any compile errors I will endeavour to help.

nicolasr75, 2017-06-22:
Ok, thanks for the tips! I will see how far I get and report back.

nzain, 2017-07-05:
If you have cmake on your arm device, you can compile it on the device.

```
cd dll/serialunix
make
make install
```

For those interested in cross compilation via buildroot, content of `Config.in`:

```
config BR2_PACKAGE_LIBNSERIAL
	bool "libnserial"
	help
		This installs the libnserial library.
```

and the `libnserial.mk` file:

```
LIBNSERIAL_VERSION = v2.x
LIBNSERIAL_SITE = $(call github,jcurl,SerialPortStream,$(LIBNSERIAL_VERSION))
LIBNSERIAL_INSTALL_STAGING = YES
LIBNSERIAL_SUBDIR = "dll/serialunix"

$(eval $(cmake-package))
```

off-topic: However, it currently doesn't show any ports. Gonna file another ticket.

nicolasr75, 2017-07-06:
@nzain Thanks, that makes it a lot easier. For anyone who is a Linux newbie like me I want to add this:

To install cmake do:
`sudo apt-get install cmake`

Copy/Download the SerialPortStream repo and navigate to the serialunix directory as advised by @nzain.
Do a `cmake .` Then do `make`. `make install` failed in my case with

```
Makefile:105: recipe for target 'install' failed
make: *** [install] Error 1
```

For my project I ignored that error and simply copied the created `libnserial.so.1.1` to my project directory and renamed it to libnserial.so.1. The next problem was that I couldn't access the serial port without root privileges. I had to add my user account to the `dialout` group

`sudo adduser <username> dialout`

After that I was able to successfully open the serial port on a Raspi with Ubuntu from a .NET Core program. Great!

jcurl, 2017-07-15:
No feedback, Closing.

closed 2017-07-15 by jcurl