jcurl/RJCP.DLL.SerialPortStream #31: How to access response data

agidden, 2017-07-07:
If we use SerialPortStream to send characters to a COM port with WriteLine, we cannot get the response sent back. 
The DataReceived event never fires and there is no data when we call Read after WriteLine, even if we wait for a few seconds.
There are no examples showing how to do this, and even the test code doesn't show how this is done.
Is it possible to get the response somehow?

jcurl, 2017-07-07:
You use it like you would the MS implementation. If you're running a 32-bit system, you can use [PortMon](https://technet.microsoft.com/en-us/sysinternals/portmon.aspx). Look at the wiki page on [Tracing](https://github.com/jcurl/SerialPortStream/wiki/Tracing) to get logs and post them here.

There's always a few things you can do and build yourself a simple Y-Cable and use a couple of USB-SER ports to monitor what's being sent and received.

If there's no Read when sending data, then it could be very well that the data being sent is not in the right format (check newlines), or your device is not sending (do you need to set any flow control?)

agidden, 2017-07-07:
Thank you for the quick response.
Since the DataReceived event is provided we assumed that it would work with a response, but it appears it does not.
The only other option is to look in the Read buffer, but it is empty.
We know the data is being sent and that a response is being returned, we just can't figure out how to get the response, or make SerialPortStream give us the response.

Looking at the MS implementation, it appears this is accomplished with a 'read' thread. Not a problem, but we were hoping to use SerialPortStream to avoid the MS issues.

jcurl, 2017-07-08:
Hi, I need you to provide a trace as I provided the link earlier. Then I might be able to see what is going wrong.

jcurl, 2017-07-15:
Any logs?

agidden, 2017-07-17:
Thank you for following up with us on this issue.

We used a serial port test program to verify we were being sent responses.
We also verified that the data we sent via SerialPortStream was being
received and acted on.

We are trying to send commands to an industrial marking device. I cannot
share any details of the implementation, including log data.

The commands are a string of 1 to 8 characters terminated with a newline.
The same string of characters terminated with a newline will be in the
response, if there was no error.

Our code used three patterns:
- Connect, send data, read response.
- Connect, launch thread to read response, send data.
- DataReceived event.

jcurl, 2017-07-17:
I'm sorry that I can't help you any further. To do so I would need logs as generated by my software using the tracing link I've provided earlier. Your issue refers to no data being received at all. This sounds like a custom driver provided by your manufacturer that doesn't fully implement the serial driver in asynchronous mode.

closed 2017-07-17 by jcurl