jcurl/RJCP.DLL.SerialPortStream #34: Does a blocking read byte method exist?

wolfen351, 2017-07-24:
Hi there

Please pardon my ignorance, but i'm looking for the simplest possible method (imho) and not finding it - a way to read a byte from the stream when it becomes available. I'm happy to block until the byte arrives. 

Please can you point me to such a method?

Thanks!

jcurl, 2017-07-24:
There is plenty that I've copied from the MSDN implementation:
* the standard `Stream.Read(byte[] buffer, int offset, int count)` method.
* the simplest `public override int ReadByte()` method.
* for chars `public int Read(char[] buffer, int offset, int count)`
* and here `public int ReadChar()`

To block forever, ensure that `ReadTimeout` is set to -1.

Cheers,
Jason.

closed 2017-07-24 by jcurl

wolfen351, 2017-07-24:
Ok, thank you. I was under the impression that these didn't block, that they returned immediately with -1 if data was not available (i'm thinking specificially about ReadByte() here). I found the wiki in the meantime, so all is good :)

jcurl, 2017-07-24:
Looking at the code:

```
public override int ReadByte()
        {
            if (IsDisposed) throw new ObjectDisposedException("SerialPortStream");
            if (m_Buffer == null) return -1;
            if (ReadCheckDeviceError()) return -1;

            if (m_NativeSerial.IsRunning) {
                if (!m_Buffer.Stream.WaitForRead(m_ReadTimeout)) {
                    ReadCheckDeviceError();
                    return -1;
                }
            }
            int value = m_Buffer.Stream.ReadByte();
            if (value != -1) m_ReadTo.Reset(false);
            return value;
        }
```

the text `m_Buffer.Stream.WaitForRead(m_ReadTimeout)` should mean it will block until data is available.