jcurl/RJCP.DLL.SerialPortStream #36: Building libserial on macOS

erndev, 2017-09-13:
Hi

I've just tested the library using macOS and Net Core 2.0. 
Just preliminar, but seems to be working fine so far.  Port enumerations (GetPortDescriptions/GetPortNames) don't seem to work, but reading/writing are working fine so far.

Just made some changes in order to build  the dll  on the mac, and i am writing them here, just in case anyone is interested. 

**Disclaimer**: i was focusing on building the serial library and had some problems building gtest, so i removed the tests and doxygen building . I'll try to get them to build when i have some free time. So, i don't recommend this procedure unless you know the risks of using an untested dll.

These are the steps.

1.  open the cmake build file (./dll/serialunix/libnserial/CMakeLists.txt) and make these changes:

- don't build doxygen. To to that, Comment out or remove this line:
  `include(${CMAKE_MODULE_PATH}/doxygen.cmake) `
- don't build the tests. Remove or comment out these two lines :
 ```
add_subdirectory(unittest)
add_subdirectory(comptest)
```
- Change the script to generate a dylib instead of a framework. Look for this line:
`FRAMEWORK ON`
and replace it whit this:
`FRAMEWORK OFF`

That's it. Now you can build it.  The include and library files will be located in the subfolder `./bin/usr/local`

Now, you can keep them there or copy then to your preferred location. Take into account that you need to include the path to the libraries in LD_LIBRARY_PATH so that your app can find them.

One last step. The source code is looking for a library called `libnserial.so.1`  , so you need to rename the library or create a symbolic link with that name. In my case, i installed the library in /usr/local/lib, an crated the link like this:

`ln -s  /usr/local/lib/libnserial.dylib /usr/local/lib/libnserial.so.1.dylib`

That's it. Did not test it much, but at least i got to send/receive data using .Net core on a mac, which is great.

thanks for this awesome piece of software!!

erndev, 2017-09-14:
@jcurl 

I think that with a small change , the build script can work in mac as well as linux.

```
if (${CMAKE_SYSTEM_NAME} MATCHES "Darwin")  
  FRAMEWORK OFF 
else  
  FRAMEWORK ON 
endif()
```
It just disables the framework if built on macos and keep it as it was in other systems.
would you accept a pull request with this change?

jcurl, 2017-09-14:
> @jcurl <https://github.com/jcurl>
>
> I think that with a small change , the build script can work in mac as
> well as linux.
>
> |if (${CMAKE_SYSTEM_NAME} MATCHES "Darwin") FRAMEWORK OFF else
> FRAMEWORK [...]
Would do for this simple change. I will only be able to test on Linux as 
I don't have a mac (and I'm not aware of any useful emulation software 
to test on in its absence).

jcurl, 2017-09-14:
Hello,

>
> Hi
>
> I've just tested the library using macOS and Net Core 2.0.
> Just preliminar, but seems to be working fine so far. Port 
> enumerations (GetPortDescriptions/GetPortNames) don't seem to work, 
> but reading/writing are working fine so far.
I don't have a mac to write such software, nor do I know how a mac 
organises its devices and how we can tell if such devices are really 
serial ports (or terminal devices). I'd be more than happy to accept a 
patchset/pull request on v2.x, with prior discussion on the organisation.
> Just made some changes in order to build the dll on the mac, and i am
> writing them here, just in case anyone is interested.
>
> *Disclaimer*: i was focusing on building the serial library and had [...]
If you know how to properly get CMake to handle all this stuff, i can 
test on Linux and integrate the mac changes.

jcurl, 2017-09-26:
If you want to prepare and build a library for a release, I can add it to the release page of GitHub, but I can't maintain it. As I understand, it's not compilable at the moment, so I need someone's branch, or patches to integrate, to ensure that whatever is released is based on the repository. Just a feature branch with the fixes is sufficient (I usually rework them before committing to keep coding styles, etc.)

andymac4182, 2017-09-29:
@erndev Do you have some example code for mac and a copy of your compiled dylib?