jcurl/RJCP.DLL.SerialPortStream #37: Open port error

MartinStokelj, 2017-10-05:
I get error on open port.
Exception thrown: 'System.IO.IOException' in RJCP.SerialPortStream.dll
Additional information: Unknown error 0x79: COM7    

Sometimes also Unknown error 0x490: COM7   

[image]

It helps only if i resrtart bluetooth. Can this be fixed somehow ?

jcurl, 2017-10-05:
I've heard of a similar issue before with bluetooth. This is being returned by the Windows itself and there is no known workaround that I know of.

Have you tried with other terminal software? Do you notice if the Bluetooth device is connected or not?

The error that you observe is raised immediately after trying to open. So if Windows won't open it, I don't know what else can be done other than your application check for any Bluetooth specific features that might be needed prior:

```
            m_ComPortHandle = UnsafeNativeMethods.CreateFile(@"\\.\" + PortName,
                NativeMethods.FileAccess.GENERIC_READ | NativeMethods.FileAccess.GENERIC_WRITE,
                NativeMethods.FileShare.FILE_SHARE_NONE,
                IntPtr.Zero,
                NativeMethods.CreationDisposition.OPEN_EXISTING,
                NativeMethods.FileAttributes.FILE_FLAG_OVERLAPPED,
                IntPtr.Zero);
            if (m_ComPortHandle.IsInvalid) WinIOError();
```

closed 2017-10-05 by jcurl

MartinStokelj, 2017-10-05:
If i look under windows bluetooth device i see state "familiar". (but this does not change until open is actualy made)
If i close bluetooth connection on windows it does not help.
If i close bluetooth connection on the device (antena) it helps. SerialPortStream is able to open port than.

It is possible to reset/restart bluetooth device some how to workaround ?

jcurl, 2017-10-05:
Hi MartinStokelj,

Nothing that the SerialPortStream can do. It sounds like a state of the bluetooth driver or remote device that is waiting for a connection to be closed. Is there a command that your device requires before closing the port locally? Any forums on the web that state a similar issue with your chipset/device?

MartinStokelj, 2017-10-05:
I hope this can help

First one works always, no need to restart
Device BTHENUM\Dev_00126F357E35\7&1ebe74a&0&BluetoothDevice_00126F357E35 
Device BTHENUM\Dev_00126FD20503\7&1ebe74a&0&BluetoothDevice_00126FD20503
Device BTHENUM\Dev_00126FD20972\7&1ebe74a&0&BluetoothDevice_00126FD20972

jcurl, 2017-10-05:
Sorry - I don't quite understand what you mean. Is that a COM port? Can you explain?

In any case, it would be outside of the scope of the SerialPortStream to handle Bluetooth devices explicitly, as it's also outside of the scope of the MS implementation of SerialPort.

MartinStokelj, 2017-10-05:
Those were bluetooth device names, i tryed 3 of them.
What is the corect way to close port in i close application ? So it can be open again when i start app ?

jcurl, 2017-10-05:
To close the SerialPortStream, you should Dispose() of it, or Close() it. Closing the port with Close() will close the device, but any buffers read can still be read, but writing is obviously no longer possible. Disposing will also close the serial port, but also discards any open buffers.