jcurl/RJCP.DLL.SerialPortStream #39: Exception when writing data to serial port und raspberry pi

p3root, 2017-10-26:
Hey!

I have no idea but I get an exception if I want to write data to the serial port. Could you help me please?

`Calling 
await _stream.WriteAsync(buffer, 0, buffer.Length);
        

Unhandled Exception: System.Exception: Error 5
   at RJCP.IO.Ports.Native.UnixNativeSerial.ThrowException()
   at RJCP.IO.Ports.Native.UnixNativeSerial.SetPortSettings()
   at RJCP.IO.Ports.SerialPortStream.Open(Boolean setCommState)
   at RJCP.IO.Ports.SerialPortStream.Open()
System.NotSupportedException: Stream does not support writing.
   at System.IO.__Error.WriteNotSupported()
   at System.IO.Stream.BeginWriteInternal(Byte[] buffer, Int32 offset, Int32 count, AsyncCallback callback, Object state, Boolean serializeAsynchronously, Boolean apm)
   at System.IO.Stream.BeginEndWriteAsync(Byte[] buffer, Int32 offset, Int32 count)
   at System.IO.Stream.WriteAsync(Byte[] buffer, Int32 offset, Int32 count, CancellationToken cancellationToken)
   at System.IO.Stream.WriteAsync(Byte[] buffer, Int32 offset, Int32 count)
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.GetResult()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.GetResult()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.AsyncMethodBuilderCore.<>c.<ThrowAsync>b__6_1(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.WaitCallback_Context(Object state)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
`

p3root, 2017-10-26:
I have changed the async functions to the normal functions. Now this works :) but I get an exception when I want to discar the buffer:

`Unhandled Exception: System.NotImplementedException: The method or operation is not implemented.
   at RJCP.IO.Ports.Native.UnixNativeSerial.DiscardInBuffer()
   at RJCP.IO.Ports.SerialPortStream.DiscardInBuffer()
`

I will take a look mysef the next days but I guess you could fix it faster than me :)

p3

jcurl, 2017-10-27:
Hi, the NotImplementedException was reported already and fixed here: https://github.com/jcurl/SerialPortStream/commit/891452df0d7954bd21f6e91ddd5e6e7d66aaaba8

Please try a newer version, relesaed here: https://github.com/jcurl/SerialPortStream/releases/tag/release%2F2.1.2.0

jcurl, 2017-10-27:
For your first exception, I need more time to investigate what is going on. I assume you're using Mono on Unix, you didn't say.

p3root, 2017-10-27:
Ok. I loaded the lib from nuget. Therefore nuget not up to date??

p3root, 2017-10-27:
1st exception: I use .NET Core on a raspberry by with raspbian stretch.

p3

p3root, 2017-10-27:
Hey again.
I've tried it with a current version of your lib. Works great, thanks so far for your work :) but a new nuget version would be awesome :)

closed 2018-04-18 by p3root

alexdresko, 2018-05-18:
To anyone who has used this library with .NET Core, Raspberry Pi 3, Raspian Stretch, what do you specify for the port name? Did you have to do anything to enable the port? I'm trying to use one of the USB ports, not the serial headers.

jcurl, 2018-05-20:
You need the native library libnserial.so, otherwise you probably get DllNotFoundException or PlatformNotSupportedException. Ensure that the user running is in the group to access /dev/ttyUSB0. The GetPortNames function should give a list of all available ports. If none is there, the most common mistake is unix permissions prevent you from opening the port.

dabasejumper, 2018-11-08:
This seems to by my issue as well...

I'm running .NET Core 2.1.4 in a docker container... Is there a way to grant access to the image running that would have access to the USB port at /dev/ttyUSB0.