jcurl/RJCP.DLL.SerialPortStream #43: Receiving 00 Bytes on Linux-ARM

3dGrabber, 2017-11-27:
I'm trying to use [this](https://www.victronenergy.com/live/open_source:ccgx:start) device as a Modbus master. Communication seems to work partially: for each request I receive a response with the expected number of bytes. However all those bytes are all `00`s!

	Tx: 02-04-03-E7-00-35-80-5D
	
	// expecting a 332 Bytes response
	
	Rx: 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
	LEN: 332

(The slave answers correctly when queried from another device)

3dGrabber, 2017-11-27:
Just recompiled with LOG enabled,   
did not find anything suspicious though:

	2017.11.27-12:04:35.113197      6       /dev/ttyUSB0    open: succeeded
	2017.11.27-12:04:35.120704      7       /dev/ttyUSB0    setproperties: getting attributes
	2017.11.27-12:04:35.121406      7       /dev/ttyUSB0    setproperties: setting attributes
	2017.11.27-12:04:35.123604      7       /dev/ttyUSB0    setproperties: setting attributes done

3dGrabber, 2017-11-27:
closing this because I suspect that it is a driver issue.   
Sorry for the confusion

closed 2017-11-27 by 3dGrabber

jcurl, 2017-11-27:
Sounds like the problem I documented on the wiki. https://github.com/jcurl/SerialPortStream/wiki/Driver-Compatibility-with-SerialPortStream#Garbage_Data_on_Start