jcurl/RJCP.DLL.SerialPortStream #44: Device Error

overthrowrobotics, 2017-12-08:
I'm using an arduino style device (Teensy 3.5) at 115200 on COM3. I can open it just find with the Arduino monitor and it just prints an incrementing number on new lines (1, 2, 3, etc). Am I missing something really simple here? 

IOException: Device Error
RJCP.IO.Ports.SerialPortStream.ReadCheckDeviceError (System.Boolean immediate) (at <98240c6c7bfd40d18eca5b369f29ae84>:0)
RJCP.IO.Ports.SerialPortStream.ReadTo (System.String text) (at <98240c6c7bfd40d18eca5b369f29ae84>:0)
RJCP.IO.Ports.SerialPortStream.ReadLine () (at <98240c6c7bfd40d18eca5b369f29ae84>:0)
(wrapper remoting-invoke-with-check) RJCP.IO.Ports.SerialPortStream:ReadLine ()
serial.Start () (at Assets/serial.cs:24)

I'm doing this from Unity 2017 running .NET 4.6. 

using RJCP.IO.Ports;

SerialPortStream src = new SerialPortStream("COM3", 115200);
// This method runs once at startup.
void Start () {

        src.Open();
        string srcString = src.ToString();
            
        Debug.Log(srcString); // same as Console.Write
        Debug.Log(src.ReadLine().ToString());
  }
 
// this runs once per frame until the program exits.
void Update() {
        Debug.Log(src.ReadLine().ToString());
}

jcurl, 2017-12-08:
The problem is that there was an unexpected error while reading from the serial port and the reader/writer thread has died. That's why you get the exception. At the very least, this page on [tracing from the wiki](https://github.com/jcurl/SerialPortStream/wiki/Tracing) will help.

There it should be possible to determine why the thread died. Minimum level is warning, verbose if you can.

Speculating (I have no idea what a teensy 3.5 is), but I would image it's a USB connection. Check that the USB connection is stable. If there was an error reported by the OS, the software generally doesn't know how to retry and it returns an error.

pigrew, 2017-12-19:
@jcurl, relating to your speculation, I'm trying to handle disconnection events in my code.  Upon disconnect, I see the serial port thread exiting, but I don't see any way to receive a callback when this event happens. Is the proper thing for me to do to poll the "CanWrite" property in a background thread? Could the ErrorReceived callback be extended to be called when a disconnect event happens?

(In my application, I very rarely write to the serial port; I'm mostly just logging status updates, but I want to warn the used if a disconnection happens.)

jcurl, 2017-12-19:
The SerialPortStream cannot know why the device reported an error and the error code that is returned may differ depending on the operating system. It was never designed this way, as a serial port may be emulated, physical that is not hotpluggable or hotpluggable. Important is on the Read method, you finally get 0 that indicates that the stream is closed. A second read results in an exception. Basically, it's the 0 from reading for a blocking connection that indicates no data is available.

If you need to know if a USB device is removed, you will have to refer to the documentation for the USB subsystem to determine hotplugging events. The Serial subsystem simply doesn't provide that information and this is not the scope of the SerialPortStream.

closed 2017-12-19 by jcurl