jcurl/RJCP.DLL.SerialPortStream #46: Including SerialPortStream in Visual Studio Installer Setup Project

ohjohnsen, 2018-01-17:
I'm trying to make a Setup Project using Visual Studio Installer in VS2015, with a project where I'm using SerialPortStream.  But when building the setup project, I get the following warning message: 

```
------ Starting pre-build validation for project 'PoseidonSetup' ------ 
WARNING: Unable to find dependency 'MONO.POSIX' (Signature='0738EB9F132ED756' Version='4.0.0.0') of assembly 'RJCP.SerialPortStream.dll'
WARNING: Unable to find dependency 'MONO.POSIX' (Signature='0738EB9F132ED756' Version='4.0.0.0') of assembly 'RJCP.SerialPortStream.dll'
WARNING: Unable to find dependency 'MONO.POSIX' (Signature='0738EB9F132ED756' Version='4.0.0.0') of assembly 'RJCP.SerialPortStream.dll'
------ Pre-build validation for project 'PoseidonSetup' completed ------
```

Do you know what this is?  I'm not manually adding any dependencies, and RJCP.SerialPortStream.dll gets added automatically.

jcurl, 2018-01-17:
The Mono.Posix library is the package for providing posix support under linux. On Windows, you don't need to include this library (it's in the dependencies of NuGet, so the NuGet package is downloaded to satisfy compile time dependencies, even though it's not used at runtime on Windows).

On Linux it isn't needed because the Mono implementation provides this in its GAC.

closed 2018-01-17 by jcurl

ohjohnsen, 2018-01-19:
Thank you for the info.  So what can be done with the warning message?  Do I just ignore the warning message when building the setup project, or can I disable it (how?)?

jcurl, 2018-01-25:
Sorry - I don't support the VS2015 installer. You should just be able to ignore the warnings, or tell the installer script somehow these files aren't needed.