jcurl/RJCP.DLL.SerialPortStream #48: FileNotFoundException when calling GetPortNames()

NilesDavis, 2018-02-14:
Hi! I am running a UWP-App to test the Library. The first step in opening a new port was to check if it is available. But when I called `SerialPortStream.GetPortNames()` the FileNotFoundException was thrown with the description: "Could not load file or assembly 'Microsoft.Win32.Registry, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'"
OK, without validating the existence of the port I wanted to connect via
```
try
{
    string[] ports = SerialPortStream.GetPortNames();
    SerialPortStream sps = new SerialPortStream("COM3", 57600, 8, Parity.None, StopBits.One);
    sps.Open();
}
catch(Exception ex)
{
    Debugger.Break();
}
```
This led to "Access Denied: COM3"
There is no problem accessing the port with other tools like HTerm, and unlike #11 there is definitely no other program talking with the port, and there is no possiblilty to have an error like #38 because the above lines of code are the only ones existing. So I downloaded and referenced the source and the error happens here:
```
m_ComPortHandle = UnsafeNativeMethods.CreateFile(@"\\.\" + PortName,
                NativeMethods.FileAccess.GENERIC_READ | NativeMethods.FileAccess.GENERIC_WRITE,
                NativeMethods.FileShare.FILE_SHARE_NONE,
                IntPtr.Zero,
                NativeMethods.CreationDisposition.OPEN_EXISTING,
                NativeMethods.FileAttributes.FILE_FLAG_OVERLAPPED,
                IntPtr.Zero);
```
as the returned handle is invalid (-1) and the `WinIOError()` @ `int e = Marshal.GetLastWin32Error();` shows a 5 leading to UnauthorizedAccessException.
Do you have a hint where to search for a possible mistakes?

jcurl, 2018-02-14:
Works fine under Win32 Desktop. I suspect UWP is started in a sandboxed environment preventing you from accessing the COM port. Error 5 comes back due to permissions issues, or someone else already has the COM port opened.

closed 2018-02-14 by jcurl

NilesDavis, 2018-02-14:
Obiously I forgot to define the device capabilities in the Package.appxmanifest like this:
```
<DeviceCapability Name="serialcommunication">
  <Device Id="any">
    <Function Type="name:serialPort"/>
  </Device>
</DeviceCapability>
```

jcurl, 2018-02-14:
Good news! If you have success, let us know your experience, and perhaps you can make some suggestions for the Wiki.

jasells, 2018-11-08:
@NilesDavis 

Did you ever resolve this?  I am looking into this as well, but I don't think this lib is compatible with UWP...  I keep getting "FileNotfound" (from GetPortNames) or UnauthorizedAccess (from Open()).

The short version is that UWP does have sandboxing in place for lots of things, including File, Network, and Serial (device) IO and while UWP does support the .NET 2.0 API's, there's no guarantee that they will work.  System.IO.File API's are similar on UWP (available, but throw runtime exceptions when called because sandboxing prohibits the low-level calls).

I know my serial device works, as I've tested it via the permitted API: Windows.Devices.SerialCommunication.SerialDevice  (etc).

In case anyone needs an immediate UWP solution, here is an [example](http://www.iot-developer.net/windows-iot/uwp-programming-in-c/uart-serial-port/rpi-using-the-uart). 

@jcurl  If you have any desire for supporting UWP, let me know.  It would be possible...