jcurl/RJCP.DLL.SerialPortStream #51: Can not get port name correctly

v-liujxu, 2018-03-16:
I used SerailPortStreamCore package to enumerate the ports on Windows IoT Core on my Raspberry Pi 3. The demo is a console app in dotnet core 2.0 (published for win10-arm).
There is a Silicon Labs CP210x USB to UART Bridge plugged in the device.
But when i run the console app, the port name shows garbled. The result is same even though there is no any USB 2 UART controller plugged in.

dotnetcore version: 2.0
windows iot core:16299

jcurl, 2018-03-20:
Hi, this indicates a bug in the IoT platform. The code that is being run is simple on Windows, to get the list of serial devices from the registry. The code is called `GetPortNames()` in the file WinNativeSerial.cs. Else under .NET 1.5 there's the file code/netstandard/System/IO/Ports/SerialPrt.cs where the function returns a zero length array. So it's hard to see, unless there's a bug somewhere in the OS or Drivers, how this component can cause this.

I'm going to close the topic. If you have the ability to debug and find the problem, please do reopen and provide possibly a patch.

closed 2018-03-20 by jcurl

jcurl, 2018-12-06:
You might want to try the newest version released, I did identify a bug with obtaining the port names. Perhaps it's now fixed.