jcurl/RJCP.DLL.SerialPortStream #54: Does it works in Visual Studio for MAC

midiway, 2018-05-13:
That is, does is works on OSX?

jcurl, 2018-05-13:
It might. There are a few tickets (see the ones that are closed), in particular one for CMake (but I wasn't 100% sure, as there was no diff provided by the reporter).

You'll need to compile libnserial for your machine, install it in your library path, and it should work. If not, provide details of what doesn't work, and it might be debugged on a separate branch until it works.

I *don't* have a mac, I have no need for a mac, there isn't really any emulation platforms that I can use without spending significant efforts getting it to work. The .NET libraries should work without modification as I expect you will use Mono.

midiway, 2018-05-13:
So, just to be clear, for OSX, this .NET lib has this libnserial dependency?
Or is it possible to partially use it without libnserial?

tmatthey, 2018-05-13:
Hi,
My 2 cents. I did a fork and replaced the all sln files by one with multiple targets. I would highly recommend it; you can then also easily add new targets. https://github.com/tmatthey/SerialPortStream

midiway, 2018-05-13:
No idea what you mean....
Anyway, I am sticking with default MONO SerialPort

closed 2018-05-13 by midiway