jcurl/RJCP.DLL.SerialPortStream #61: cmake comptest

pigwing, 2018-07-11:
hi i try to cmake the comptest dir file and throw some exception

 Building C object CMakeFiles/icount.dir/icount.o
/var/serialportstream/dll/serialunix/libnserial/comptest/icount.c:1:20: fatal error: config.h: No such file or directory
compilation terminated.
CMakeFiles/icount.dir/build.make:62: recipe for target 'CMakeFiles/icount.dir/icount.o' failed
make[2]: *** [CMakeFiles/icount.dir/icount.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/icount.dir/all' failed
make[1]: *** [CMakeFiles/icount.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

jcurl, 2018-07-17:
Please provide a more detailed description of how you reproduce the error. Naturally on my machine I can't recreate the issue. Best would be to attach console output, exactly what you're compiling, the system you're compiling on.

jcurl, 2018-08-17:
Didn't get any more details, and works find on my Ubuntu 18.04 with HEAD of v2.x

closed 2018-08-17 by jcurl