jcurl/RJCP.DLL.SerialPortStream #62: RtsEnable depends on port being open?

onionhammer, 2018-07-12:
If setting 'RtsEnable' on nix, a bad file descriptor error occurs

```C#
 var port = new SerialPortStream(portOptions.Path)
            {
                BaudRate     = (int)portOptions.BaudRate,
                DataBits     = portOptions.DataBits,
                Parity       = ConvertParity(portOptions.Parity),
                StopBits     = ConvertStopBits(portOptions.StopBits),
               /// RtsEnable  = true/false,  /// <--- Causes crash "bad file descriptor"
                ReadTimeout  = portOptions.ReadTimeoutMs,
                WriteTimeout = portOptions.WriteTimeoutMs
            };
```

jcurl, 2018-07-17:
Thanks for the report. Can you say what version of libnserial that you're using?

onionhammer, 2018-07-17:
@jcurl I built it from source / master

jcurl, 2018-08-17:
Implemented in HEAD.

closed 2018-08-17 by jcurl