jcurl/RJCP.DLL.SerialPortStream #67: Close two serial ports hang on MacOS

howlfu, 2018-08-28:
Hi jcurl,
Here is my code use dotnet core 2.1:
```SerialPortStream tmp1 = new SerialPortStream("/dev/tty.usbserial");
tmp1.Open();  
SerialPortStream tmp2 = new SerialPortStream("/dev/tty.usbserial2");
tmp2.Open();
Thread.Sleep(3000);
tmp1.Close();
tmp2.Close();```

And it hangs when tmp2 close.
tmp2.Close();

I tried to trace code it stop when call C library,
```
[DllImport("libnserial.so.1", SetLastError = true)]
        internal static extern int serial_close(SafeSerialHandle handle);
```
But it goes well on linux, any idea?
Thanks.

jcurl, 2018-08-28:
Hi, you can in more detail exactly where it hangs if you recompile with logging enabled.

```
CFLAGS="-O0 -g -Wall" cmake -DCMAKE_BUILD_TYPE=Debug -DNSLOG_ENABLED=ON .. && make
```

Then the log file will say on exactly which system call it is hanging. Please do that and provide logs. I don't have MacOS to test

jcurl, 2018-08-29:
Thankyou. The problem can occur due to one of two APIs. Can you please send me your `config.h` file?

The location where it occurs is in this path, either the `ioctl` or the `close`.

```c
#if defined HAVE_TERMIOS_EXCLUSIVE
  if (ioctl(handle->fd, TIOCNXCL)) {
    nslog(handle, NSLOG_NOTICE, "close: error setting TIOCNXCL: errno=%d", errno);
  }
#endif

result = close(handle->fd);
```

Also, does the problem only occur with `/dev/tty.usbserial16`? If you were to swap the ports would it work? I've seen some bluetooth adapters do strange things.

Can you also provide for reference the serial hardware you're using?

howlfu, 2018-08-29:
Thanks for your prompt reply.
Here is config.h. 
[config.h.zip](https://github.com/jcurl/SerialPortStream/files/2331178/config.h.zip)

This problem occur when you close second port, no matter /dev/tty.usbserial16 or /dev/tty.usbserial.
Swap the ports still hang. But when only plug one port to MacOS, it works normally.

The reference hardware:
1. LoRa module to USB.
https://www.microchip.com/wwwproducts/en/RN2903
2. NBIoT usb device

jcurl, 2018-08-29:
OK. The MacOS-X says it supports TIONCXL. If you were to modify the config.h to remove the defines and recompile, does it work?

Remove these lines please from config.h and execute make again. This will tell me if the drivers in MacOS 
```c
#define HAVE_TERMIOS_TIOCNXCL
#define HAVE_TERMIOS_TIOCEXCL
#if defined(HAVE_TERMIOS_TIOCEXCL) && defined(HAVE_TERMIOS_TIOCNXCL)
#define HAVE_TERMIOS_EXCLUSIVE
#endif
```

This tells the system that opening and closing the drivers should have exclusive access (i.e. while open, a second open should fail).

If it works, either you are opening the same serial port twice, once in .NET, another somewhere in a different application, or MacOS is buggy and doesn't behave as expected.

jcurl, 2018-09-11:
Hi, do you have feedback if removing TIOCNXL works?

jcurl, 2018-10-12:
As I have no MAC to test, could you please confirm if TIOCNXL is the problem?

closed 2018-11-16 by jcurl