jcurl/RJCP.DLL.SerialPortStream #69: Incorrect IsOpen value

VPeruS, 2018-09-07:
After device manual disconnection IsOpen property of SerialPortStream continue to return true.
We using USB to COM converted, host system Windows 7.

Code that was used for testing
```csharp
        static void TestRJCP()
        {
            var s = new SerialPortStream("COM4");

            s.Open();

            while (true)
            {
                foreach (var portName in SerialPortStream.GetPortNames())
                {
                    System.Console.Write($"{portName} ");
                }
                System.Console.WriteLine();

                System.Console.WriteLine($"{s.IsOpen}");
                if (s.IsOpen)
                {
                    try
                    {
                        s.WriteByte(1);
                    }
                    catch (Exception ex)
                    {
                        System.Console.WriteLine($"{ex}");
                    }
                }
            }
        }
```

After disconnect s.WriteByte will continuously throw exception

jcurl, 2018-09-09:
Thank you for the report. Even if the underlying device driver is removed, the object still needs to be closed to free up resources - the library doesn't do this automatically (that would lead to memory errors). Have you tested the MS implementation? If this differs to MS, then I'll certainly consider a change to improve compatibility

VPeruS, 2018-09-09:
Yes, it different from MS's, IsOpen starts to returns false after device removed.

jcurl, 2018-09-09:
Thankyou. I'll consider this a bug.

jcurl, 2018-11-12:
Please note, that the IsOpen now returns false when the device is physically removed, but the SerialPortStream still needs to be closed. Resources are still being held.

closed 2018-11-12 by jcurl