jcurl/RJCP.DLL.SerialPortStream #70: SerialPortStream.ReadLine() stops reading data

bildsee, 2018-09-11:
My software reads from a device that permanently sends ASCII lines (CR+LF) at 4800,8N1 no flow control.
So it basically executes a loop calling SerialPortStream.ReadLine() which works fine most of the time.
However, sometimes ReadLine() fails and throws TimeoutException. This may be caused by the device and is not the actual problem. Is is also no problem if some lines get lost. The problem is that afterwards ReadLine() **_never_** returns data but always throws TimeoutException although the device continues to send (btw, the read timeout = 500ms). I found a workaround which 'fixes the stream' by reading some bytes via SerialPortStream.ReadBytes(...) in case of TimeoutException. This call returns data and afterwards ReadLine() works as expected again. 
So I'd expect ReadLine() to return the next available line after it timed out.

jcurl, 2018-09-11:
Thankyou. To get further to the root cause of the problem, could you please provide .NET traces enabling the ReadTo and the serial port traces? This may help me to identify where it's getting blocked.

jcurl, 2018-09-25:
Hello, are you able to capture .NET traces of the error and attach them to the ticket?

bildsee, 2018-09-26:
Thanks for your dedicate reply. I did not yet find the time to do so but I will check as soon as possible.

jcurl, 2018-10-12:
Hello, have you found the time since reporting the issue to reproduce and to capture logs?

rquackenbush, 2018-11-01:
My team is running into a similar problem. After a while (usually weeks),  our process stops reading from devices. We're working on a repro, but it's difficult given how long it takes for the issue to manifest. Will report back as soon as we have something.

rwinscot, 2018-12-17:
Experiencing a long-running issue on Linux as well... seems to work fine and then several days later reads start failing.

2018-12-17 17:11:26.3045 Error reading from serial port: System.ArgumentException: offset and count exceed buffer boundaries
   at RJCP.IO.Ports.SerialPortStream.ReadCheck(Byte[] buffer, Int32 offset, Int32 count)
   at RJCP.IO.Ports.SerialPortStream.Read(Byte[] buffer, Int32 offset, Int32 count

jcurl, 2018-12-18:
Do you have logs captured that you can post?

closed 2021-10-14 by jcurl