jcurl/RJCP.DLL.SerialPortStream #78: System.IO.IOException: DeviceError on Linux

geversmann, 2018-11-13:
We are using 
-          Linux 4.12.4-1 Arch
-          Libnserial.so.1.1
-          SerialPortStream 2.1.4.0
-          3 COM ports (9600, 38400, 57600 baud)
-          .NET Core 2.1

We open the ports once.
Then a short time for 8 min all is running well. 
But then we get following errors at SerialPortStream.ReadExisting()/.Write():

System.IO.IOException: Device Error
   at RJCP.IO.Ports.SerialPortStream.ReadCheckDeviceError(Boolean immediate) in C:\EASE\MetaBridge_Core\Brabender.Common.Hardware5.SerialPortStream\SerialPortStream.cs:line 630
   at RJCP.IO.Ports.SerialPortStream.ReadExisting() in C:\EASE\MetaBridge_Core\Brabender.Common.Hardware5.SerialPortStream\SerialPortStream.cs:line 932
   at Brabender.Common.Hardware5.BrabSerialPort.SerialPort.ReadExisting() in C:\EASE\MetaBridge_Core\Brabender.Common.Hardware5.BrabSerialPort\SerialPort.cs:line 502

or

System.IO.IOException: Device Error
   at RJCP.IO.Ports.SerialPortStream.WriteCheckDeviceError() in C:\EASE\MetaBridge_Core\Brabender.Common.Hardware5.SerialPortStream\SerialPortStream.cs:line 1101
   at RJCP.IO.Ports.SerialPortStream.WriteCheck(Byte[] buffer, Int32 offset, Int32 count) in C:\EASE\MetaBridge_Core\Brabender.Common.Hardware5.SerialPortStream\SerialPortStream.cs:line 1091
   at RJCP.IO.Ports.SerialPortStream.Write(Byte[] buffer, Int32 offset, Int32 count) in C:\EASE\MetaBridge_Core\Brabender.Common.Hardware5.SerialPortStream\SerialPortStream.cs:line 1139
   at Brabender.Common.Hardware5.BrabSerialPort.SerialPort.Write(Byte[] buffer, Int32 offset, Int32 count) in C:\EASE\MetaBridge_Core\Brabender.Common.Hardware5.BrabSerialPort\SerialPort.cs:line 397

When we enable tracing then the problem disappears. 

So we have debug the problem by a remote session:
[image]
[image]
[image]

jcurl, 2018-11-16:
the device error indicates that the I/O thread has stopped due to an I/O error earlier. To debug this, you need to capture traces (preferably verbose, if not, then info) with the information as provided in the wiki. See [Tracing](https://github.com/jcurl/SerialPortStream/wiki/Tracing). To get to the root cause, it may be necessary to recreate also with "strace" on your system, which shows the results of all system calls.

When providing information about the libnserial, did you compile it yourself, or use a .deb package I've already provided? If you compiled it yourself, the commit ID is useful.

geversmann, 2018-11-21:
As mentioned, we have already enabled tracing.
We do this in code as following because .NET Core doesn’t supports app.config anymore.
[image]
But if we run our application with enabled tracing we don’t get the error anymore.

So we have debugged the problem as already shown.
In case of the problem we get an error result at the call “m_Dll.serial_read(m_Handle, (IntPtr)bo, length);”
The error code is 5 and the message is “Read-End-Of-File reached”.

**Update:
When we enable tracing and starting remote debugging then the problem appears too.
So we think it is a timing Problem.**
[1120c_RJCP.log](https://github.com/jcurl/SerialPortStream/files/2603964/1120c_RJCP.log)

jcurl, 2018-11-23:
Hello, are you able to track down if this is a device issue? To confirm, an `strace` is needed at the time of failure.

geversmann, 2018-11-23:
The strace will take some time because our Linux expert is currently not present.

We use multithreaded access to SerialStreamPort, but to different ports. 
The accesses to each port are already locked at high level. 
By trial, I have locked each SerialStreamPort call with a static object, without success.

In the log there is always an “**abortwatiforevent**” before the error appears:

IO.Ports.SerialPortStream Verbose: 0 : /dev/ttyBrab_Bb9ur5o6jsg10_0: ReadWriteThread: serial_waitforevent(RJCP.IO.Ports.Native.Unix.SafeSerialHandle, ReadEvent) == ReadEvent
IO.Ports.SerialPortStream Verbose: 0 : /dev/ttyBrab_Bb9ur5o6jsg10_0: ReadWriteThread: **serial_abortwaitforevent**(RJCP.IO.Ports.Native.Unix.SafeSerialHandle) = 0
IO.Ports.SerialPortStream Error: 0 : /dev/ttyBrab_Bb9ur5o6jsg10_0: ReadWriteThread: Error reading data; errno=5; description=Read End-Of-File reached

I think the **abortwaitforevent** comes from a Write call to port while the port receives characters.
Do you think this can be a problem?

jcurl, 2018-12-06:
Not sure how to continue, but at this time I believe it is a driver error, as it isn't expected to get a read of zero. If you're able to get more information, please reopen this ticket and I'll look at it again.

closed 2018-12-06 by jcurl