jcurl/RJCP.DLL.SerialPortStream #86: Missing SerialPortStream project in unit test projects

ToddMOsborne, 2019-02-26:
When I was compiling your main solution file, the unit tests failed to build before a project reference was not added to the main SerialPortStream assembly. I updated this and all is well. I also created .net 4.7.1 versions of all projects and solutions. Let me know if you are interested in having them. Thanks, works great so far!

jcurl, 2019-02-26:
Which solution file are you referring to? The `SerialPortStream-net40.sln` and `SerialPortStream-net45.sln` work for me, when opening with VS2015. The `SerialPortStream-netstandard15` loads with VS2017 and doesn't reference test cases (I don't have the time to support .NET Core, nor Mac).

jcurl, 2019-02-26:
Also, for 4.7.1, any functional changes, or just reference updates? In which case, if there are no functional changes, I prefer to support the lowest available standard, as they can then be used by any newer version of .NET allowing the greatest possible deployment.

ToddMOsborne, 2019-02-26:
I started by opening the SerialPortStream-net45.sln which include the
projects SerialPortStreamTest-net45 and DatastructuresTest-net45. One or
both of them did not have a project reference to SerialPortStream-net45, so
I added them. They would not compile as is because of the missing
reference. No functional changes for 4.7.1, just updated the project files.
Thanks,

Todd

jcurl, 2019-03-24:
I'm not able to reproduce the issue when I check out the files on my system. I don't know what is different from yours.

For now, I need to keep with .NET 4.0 as that is what I need, and there is nothing needed to require .NET 4.7.1 (with .NET 4.5 already provided, it is using whatever version of .NET you've got installed).

closed 2019-03-24 by jcurl