jcurl/RJCP.DLL.SerialPortStream #87: BytesToRead always 0

cverver, 2019-03-18:
I am trying to read data from a serial device (teensy) that constantly sends data, yet `BytesToRead()` is always 0 and trying to force a read anyway renders me an `IOException: Device Error`.
I can see the data from the serial device in putty just fine using the same COM port and baud rate, yet using this library the buffer magically remains empty.
After opening i verified the SerialPortStream is open by logging `.IsOpen`, and i check if `BytesToRead()` is greater than 0 before trying `Debug.Log(mySerialPortStream.ReadLine());` but i never see even a blank log line after IsOpen logs `true`.
I am positive that the serial device is not in use by another program and is actually sending data as fast as it can.

cverver, 2019-03-18:
I am working on Windows 10, in Unity 2019.1.0b6 and using .NET 4.x.

jcurl, 2019-03-24:
You don't say what Operating System you're using, nor what driver provides the COM port. It sounds to me like the COM port driver is incompatible. Please capture logs and post them here (instructions in the wiki). If you're using Windows, the main IO loop will exit if the driver reports an error which explains the problems you're seeing. It's unfortunate, there are a lot of poorly written drivers from 3rd parties.

The best I've found are standard 16550A from Windows, and FTDI USB.

cverver, 2019-03-25:
> You don't say what Operating System you're using, nor what driver provides the COM port. It sounds to me like the COM port driver is incompatible. Please capture logs and post them here [...]

I am using windows 10 pro x64, my COM port driver is microsoft's USBSER.SYS driver.
Arduino IDE built-in serial monitor and putty serial connection could both read the data from the device without any problem, cleanly exiting while doing so, but using this library i cannot get it to work.
The code on the teensy is simply an incrementing number printed to the serial every 10 ms or so.

In the past week i found another way to use the serial port even though it is removed in the C# version i am working with, so this issue is no longer relevant to me after 7 days.

jcurl, 2019-03-25:
My apologies, you did write the OS and the device (I misinterpreted teensy as small, not an actual device). Looking up the specs, it appears to implement the USB CDC protocol itself (it appears to be multipurpose) and probably doesn't implement the CDC USB specification completely.

That you're getting a `IOException: Device Error` is an indication that one of the MS API is giving an error and the main I/O loop is then aborting. This explains why you're not receiving data. You'll need a much simpler library.

A .NET log (modify your app.config for logging) can help to say why it crashes, but I suspect it is unlikely the problem can be solved without a rewrite of the library. Most light weight implementations usually do not implement properly the Windows API `DoWaitCommEvent`. It may work on Linux however.

I'm closing this issue as a device problem.

Good luck!

closed 2019-03-25 by jcurl

cverver, 2019-03-26:
Strange thing is that i have gotten `IOException: Device Error` in the past with different code that i've scrapped by now, usually due to some hotplug issue (MS serialport is not good with hotplugging), but now i see no errors or even warnings at all.

Using different libraries i have ran into different problems such as no DataReceived event (in an event-based serialport library).

My teensy (3.2) is currently running in Serial/Keyboard/Mouse/Joystick mode, using the built-in HID descriptor, But the way i see it those are different "channels" that don't interfere with each other.

jcurl, 2019-03-26:
Do you think you can provide a .net log? There it will give the error code from the windows API that is causing the failure.

cverver, 2019-03-26:
Would a normal unity output_log.txt suffice?
I don't have as much freedom to do unusual things or switch target framework because i am working in Unity..

jcurl, 2019-03-26:
No - it needs to be a .NET log with the changes to your "App.Config" file as documented in the Wiki.

As an example:

```
<?xml version="1.0" encoding="utf-8" ?>
<configuration>
  <system.diagnostics>
    <sources>
      <source name="IO.Ports.SerialPortStream" switchValue="Verbose">
        <listeners>
          <clear/>
          <add name="myListener"/>
        </listeners>
      </source>
    </sources> 
    <sharedListeners>
      <add name="myListener" type="System.Diagnostics.TextWriterTraceListener" initializeData="logfile.txt"/>
    </sharedListeners>
  </system.diagnostics>
</configuration>
```

Michael-Biernacki, 2024-08-06:
Having the same issue with the same specs, I know this is a dead thread, but how did you fix this?