jcurl/RJCP.DLL.SerialPortStream #90: SerialPortStream.IsOpen throws NullreferenceException

JesperNJessen, 2019-06-12:
On occasion, the IsOpen property will throw a NullreferenceException. I have not been able to reliably reproduce the issue, but it happens 1-2 times per day during normal use.

FrankHileman, 2019-06-12:
I found code in WinNativeSerial that could cause this exception. The following IsOpen code in the native property will cause problems across threads:
`return m_ComPortHandle != null && !m_ComPortHandle.IsClosed && !m_ComPortHandle.IsInvalid;`
The field can be null between the time it is checked for null and the time the IsClosed is accessed. A safer technique, for all similar properties that can be nulled:
`var handle = m_ComPortHandle; return handle != null && !handle.IsClosed && !handle.IsInvalid`
I suggest to review all nullable, mutable fields.

jcurl, 2020-05-02:
It's been a while. I'm going to modify the code so that `IsOpen` doesn't throw a `NullReferenceException`. However, there is generally no expectation that after a Close() the API will not throw an exception if not synchronized.

So that means if one thread is polling with `IsOpen` and then if the result is true, it will do something, this will not work, unless there is a critical section implemented preventing `Close()` from being called from user code while the other API is in use.

jcurl, 2020-05-06:
Released in v2.2.1 on NuGet

closed 2020-05-06 by jcurl