jcurl/RJCP.DLL.SerialPortStream #91: port.Close() hangs if 'ccTalk' setup is used

Catter38, 2019-06-28:
The issue was observed with the ccTalk protocol ([more information](https://en.wikipedia.org/wiki/CcTalk)).
TLDR: ccTalk is a protocol, where you have to short the RX and TX lines, to get a single multi-drop bus line.

If I try to close a port using this setup, it hangs and even if I force kill the application, I have to restart the computer to get it working again, otherwise I get an `System.UnauthorizedAccessException` if I try to open the port again.
I was able to trace the hang back to line 257 in `CommOverlappedIo.cs` (`m_Thread.Join();`).
It seems like some kind of deadlock in `OverlappedIoThreadMainLoop` since it never finishes the thread, even if `m_StopRunning` is set.

The only way to reproduce the issue, is to cut a serial cable and short the RX and TX lines.
After that it can easily be reproduced with something like this:
```
var port = new SerialPortStream("COM8", 9600, 8, Parity.None, StopBits.One);

port.Open();
port.DiscardInBuffer();
port.Write(new byte[] {0x00, 0x01, 0x02, 0x03}, 0, 4);

Thread.Sleep(40);

port.Close();

Console.ReadLine();
```

jcurl, 2019-06-30:
when you say "short the Rx/Tx" lines, do you mean connect RX <-> TX, or RX - GND and TX - GND?

Catter38, 2019-07-01:
I mean RX <-> TX.
This is needed to be able to connect many devices to a single "Data" line, so every device can talk to every other device on the "Bus".

jcurl, 2019-07-12:
What you've described is just nothing more than testing with a NULL modem cable. Could you further investigate by identifying at what line the software is blocked in the thread that is not joining?

Catter38, 2019-07-12:
Yes, I'll try to identify the exact line and report back if there are any news.

jcurl, 2020-05-01:
I need a reproducible case, and possibly a coredump with the binaries where the hang occurs so I can further investigate, as don't have any equipment here. Closing.

closed 2020-05-01 by jcurl