jcurl/RJCP.DLL.SerialPortStream #94: UWP Support

YZahringer, 2019-08-22:
The UWP platform is supported?

I get a error when I try to open a port with `new RJCP.IO.Ports.SerialPortStream('COM11').Open();`:
> System.UnauthorizedAccessException: 'Access Denied: COM11'
at RJCP.IO.Ports.Native.WinNativeSerial.WinIOError()
   at RJCP.IO.Ports.Native.WinNativeSerial.Open()
   at RJCP.IO.Ports.SerialPortStream.Open(Boolean setCommState)
   at RJCP.IO.Ports.SerialPortStream.Open()

No problem with `new System.IO.Ports.SerialPort('COM11').Open();`

jcurl, 2019-08-26:
You might want to look at this issue: https://github.com/jcurl/SerialPortStream/issues/48

closed 2019-08-26 by jcurl

YZahringer, 2019-08-26:
@jcurl if I understand correctly, the UWP platform is not supported and there are no plans to support it?

jcurl, 2019-08-26:
Specifically, there is no specific support for UWP from the SerialPortStream. Based on the issue ID in the previous comment, I assume that this needs to be done by your application to allow the OS to give you the permissions required.

If you identify that the change needs to be done in the SerialPortStream, you can open a PR with your test results.

YZahringer, 2019-08-26:
UWP app is sandboxed, we can't access the Win32 APIs. It is necessary to use the official APIs provided, in this case `System.IO.Ports.SerialPort`.

Thanks for the quick answer.