jcurl/RJCP.DLL.SerialPortStream #95: Serial Port already opened

guaicaiboshi, 2019-08-27:
try
                {
                    if (!this.serialPortStream.IsOpen)
                    {
                        this.serialPortStream.Open();
                    }
                }
                catch (Exception e)
                {
                    this.Logger.InfoFormat($"Open port：{this.currCommand.PortName}  failure：{e}");                    
                    return;
                }

Open port：COM5 failure：System.InvalidOperationException: Serial Port already opened.

My program ran for a while and occasionally had this problem.I'm using version 1.0.0.
I am running on Windows core2.2

jcurl, 2019-08-27:
Sorry to hear. The OS doesn't provide a mechanism to know if the serial port is in use or not, so software can only track the usage of the Open() and CloseHandle() APIs. I suspect that the OS is reporting this error, which means that internally it gets the CloseHandle() request, but doesn't clean up resources immediately. This would be driver dependent.

Do you have a backtrace of the exception? I expect it to be raised by Windows directly.

guaicaiboshi, 2019-08-29:
I need to run for a period of time again see if we can, if so, I need to add logic to handle, if it is System. InvalidOperationException: Serial Port already the opened is unusual, I will continue to implement the logic behind

closed 2020-05-01 by jcurl