jcurl/RJCP.DLL.SerialPortStream #108: serialPortStream in centOS 7

vinuthap333, 2020-10-27:
Hello,

I am trying to use serialPortStream for centOS 7. But, when I run cmake, the build is not successfull. I have followed the same steps as mentioned in the readme file. But the same is working in ubuntu 20.04.Could you please help me to achieve the same in centOS 7.

Regards,
Vinutha

jcurl, 2020-10-27:
Hi, if you can be more specific, by providing build logs, the steps you took etc. I don't have CentOS 7 installed anywhere and it's not an OS that I use. Maybe there's an invalid check. Are you prepared to make changes and help debug?

vinuthap333, 2020-10-28:
> 
> 
> Hi, if you can be more specific, by providing build logs, the steps you took etc. I don't have CentOS 7 installed anywhere and it's not an OS that I use. Maybe there's an invalid check. Are you prepared to make changes and help debug?

Hi,

I am using toolchain to cross compile serialPortStream Library in centOS 7 and the compiler used are arm-poky-linux-gcc and arm-poky-linux-g++ using toolchain. 

Steps followed

1. git clone https://github.com/libplctag/libplctag.git

2. cd libplctag
   mkdir build
   cd build

3. source /opt/............ (path to the toolchain)

4. export CC=arm-poky-linux-gnueabi-gcc
   export CXX=arm-poky-linux-gnueabi-g++

Could you please help me how to build it using these compilers.

Regards,
Vinutha

jcurl, 2021-01-08:
I can't support you with cross compilation. If you do find an error in the CMAKE rules and provide a patch, this can be fixed in the repository.

closed 2021-01-08 by jcurl