jcurl/RJCP.DLL.SerialPortStream #110: StackOverflow Exception (.net 3.5 porting )

arkypita, 2020-12-27:
Hi all

I am trying to port your library to use it in a .net 3.5 project https://github.com/arkypita/LaserGRBL
The only things I had to change are:

- add a Stream.CopyTo method (not available in 3.5)
- Handle.Dispose() => Handle.Close() (dispose not available in 3.5)
- AddrOfPinnedObject() + m_ReadBuffer.End => new IntPtr(m_ReadHandle.AddrOfPinnedObject().ToInt64() + m_ReadBuffer.End) (+ overload not available in 3.5)
- string.IsNullOrWhiteSpace(PortName) => string.IsNullOrEmpty(PortName?.Trim()) (IsNullOrWhiteSpace not available in 3.5)

Add a TestForm (demo application) to the solution

- Set destination framework 4.0 for both library and TestForm: compile and run OK
- Set destination framework 3.5 for library and 4.0 for TestForm: compile and run OK
- Set destination framework 3.5 for library and 3.5 for TestForm: compile and crash KO

Crash is a stackoverflow, cannot find where and why.

[image]

Any idea? 

Forked my project here: https://github.com/arkypita/SerialPortStream file TestSolution.sln

arkypita, 2020-12-29:
Look like is something related to the "IntPtr Add" that is not available in 3.5 and that I have solved with

AddrOfPinnedObject() + m_ReadBuffer.End => new IntPtr(m_ReadHandle.AddrOfPinnedObject().ToInt64() + m_ReadBuffer.End) 

But i cannot figure how why it crash. Maybe it is not a stack overflow but a memory/pointer corruption?
Any hint?

arkypita, 2020-12-29:
PS: if I switch from AnyCPU to x86 it work also in 3.5.
Any hint?

arkypita, 2020-12-31:
> Look like is something related to the "IntPtr Add" that is not available in 3.5 and that I have solved with
> AddrOfPinnedObject() + m_ReadBuffer.End => new IntPtr(m_ReadHandle.AddrOfPinnedObject().ToInt64() + m_ReadBuffer.End)

It is not related with this IntPtr Add .net 3.5 translation because it happens even if I simply return "m_ReadHandle.AddrOfPinnedObject()" (the buffer start address).

Sometime it was an Overflow exception, sometime an access violation, sometime a nullreference. I believe it happens inside the asynchronous code of the library, because the stack trace is almost never available (and when it is it contains only one line) and the program crash suddenly.

[image]

[image]

Trying to follow the flow of the code I saw that everything is done with a `WriteFile `whose result is read by `GetOverlappedResult`. 

If I run the code in .NET Framework 4.0 everything is fine, regardless of what target cpu is set (x86, x64, AnyCPU) all is fine.

`GetOverlappedResult `return true, and "`bytes`" variable contains the number of written byte.

If I run the code in .NET Framework 3.5 **and** if I set x64 or AnyCPU as target CPU it crash (it work only with x86).

Besides crashing, the particular thing is that `GetOverlappedResult `returns true, but the number of bytes written, returned by the  "`bytes`" variable is zero.

arkypita, 2020-12-31:
Unfortunately, this is beyond my knowledge, so I ask for help from those who know more than me.

jcurl, 2020-12-31:
As you think the problem might be related to the overlapped calls, it's probably good to start looking there in more detail. I would first assume that for some reason the memory handling with the overlapped P/Invoke isn't working as expected.

Interesting is that the buffers are pinned, so they shouldn't move around in memory. You could add trace debug where the buffers are being used by "ReadFile" and "WriteFile". Each write should be one byte or more and the actual addresses should be valid always. You could add some asserts in the code to ensure this is always the case around https://github.com/jcurl/SerialPortStream/blob/v2.x/code/Native/Windows/CommOverlappedIo.cs#L679

Also, one observation I have is the following code might be incorrect:
```csharp
[DllImport("kernel32.dll", SetLastError = true, CharSet = CharSet.Unicode)]
        [return: MarshalAs(UnmanagedType.Bool)]
        public static extern bool ReadFile(SafeFileHandle hFile, [Out] IntPtr lpBuffer, uint nNumberOfBytesToRead,
            out uint lpNumberOfBytesRead, [In] ref System.Threading.NativeOverlapped lpOverlapped);
```

Change `[Out] IntPtr lpBuffer` to `IntPtr lpBuffer`.

jcurl, 2021-01-08:
Did the proposal solve your problem?

arkypita, 2021-01-09:
Well, actually I threw in the towel.
I switched to using another code I found here: https://github.com/mono/mono/tree/master/mcs/class/System/System.IO.Ports

closed 2021-01-09 by jcurl