jcurl/RJCP.DLL.SerialPortStream #112: rts exception

maas78, 2021-01-08:
Hi, I use a rs485 serial card  b&b electronic model 3PCIOU1 on windows 10.On this card I need to change rts-control to Rs485 mode. After that I've got an exception "unable to set rts state" when trying to open port. My old vb6 program works fine with this card. How to fix this issue ? I tried 3 ways( without change any rts property(default) with your api, and trying with both rts enable/disable with same exception.

jcurl, 2021-01-08:
Perhaps you can provide a backtrace for the exception? Are you using a standard Windows driver? Or one for your card?

maas78, 2021-01-08:
I use the 3.11 version of "B & B ELECTRONICS MANUFACTURING COMPANY". My bad windows version is 7 (64 bit).
any of this useful ?. The exception happen with the Start Method.
VentureSerial.Communications.Extensions.CustomException: COM2:9600,8,E,1,to=off,xon=off,idsr=-,icts=-,odtr=-,orts=-
 ---> System.IO.IOException: Unable to set RTS state explicitly
   at RJCP.IO.Ports.Native.Windows.CommModemStatus.SetRts(Boolean value)
   at RJCP.IO.Ports.Native.WinNativeSerial.SetPortSettings()
   at RJCP.IO.Ports.SerialPortStream.Open(Boolean setCommState)
   at RJCP.IO.Ports.SerialPortStream.Open()

jcurl, 2021-01-27:
This is thrown by the SerialPortStream when it starts up, because the underlying driver reports an error. The backtrace is of course very useful. It's probably the driver throwing an exception because RS485 doesn't support flow control. The solution is to incorporate this library in your own code and to remove the exception.

```csharp
public void SetRts(bool value) {
  if (!UnsafeNativeMethods.EscapeCommFunction(m_ComPortHandle, value ? NativeMethods.ExtendedFunctions.SETRTS : NativeMethods.ExtendedFunctions.CLRRTS)) {
    throw new IOException("Unable to set RTS state explicitly", Marshal.GetLastWin32Error());
  }
}
```

jcurl, 2021-01-27:
As a workaround, you could try the `OpenDirect` method, that doesn't request to set the settings when opened. Useful for some non-wire serial drivers, and then you can try changing the values explicitly after the open.

closed 2021-03-19 by jcurl