jcurl/RJCP.DLL.SerialPortStream #120: Seeing (closed) issue "Unable to clear/set serial break state" exception in Open() once again on Windows

creature7, 2021-05-19:
Seeing this issue again, on Windows, with one device (ATEN USB to Serial Bridge), using 2.3.1:

https://github.com/jcurl/SerialPortStream/issues/2#issue-145936669

jcurl, 2021-05-23:
I don't have this device. Can you please provide logs, or a backtrace when it occurs?

jcurl, 2021-06-01:
To be able to follow this problem, it's important that I get a backtrace or more details for the issue. For this I need your help, as I don't have any ATEN USB to serial bridge devices. Thank you!

jcurl, 2021-06-04:
I'm going to close this ticket, as the ticket referenced is the commit bd134b5d45d7a5a8921e82fcc4897e1487e0982f, but was reverted in the commit 679244b99f15c07d12751ef307366132aad6d78c which shouldn't call the `SetCommBreak` or `ClearCommBreak` at all (as other systems hang). This indicates you're doing an explicit call to the property `BreakState`, which I otherwise can't tell due to lack of a backtrace at the time of the error.

closed 2021-06-04 by jcurl