jcurl/RJCP.DLL.SerialPortStream #122: Reason for calling main class SerialPortStream instead of SerialPort?

L0uisc, 2021-07-06:
I've recently started using this library to replace MS's implementation in `System.IO.Ports` and so far I'm impressed.

Just one niggle comes to mind: I have to replace all occurrences of `SerialPort` with `SerialPortStream` to port code using the MS implementation to yours. Do you have any specific reason for naming your class `SerialPortStream` instead of `SerialPort`? Your class can't collide with MS's, since both are namespaced. Furthermore, you named almost all other types and events exactly the same as MS, so there would already be collisions if you would include both namespaces via `using` statement. It would make changing simpler if it were named `SerialPort`, since then one would only need to change `using System.IO.Ports` at the top of each file to `using RJCP,IO.Ports` to port existing code.

I know it's probably too late now, since you have users who already have code using `SerialPortStream` objects. But can you please consider this if you ever release a 3.0?

bochelie1000, 2021-07-09:
Good idea!

jcurl, 2021-07-11:
All my use cases when I created this project a few years ago required a buffered stream. I didn't see the need for a direct copy of MS-API, and it was an additional overhead (developer time) that I didn't need. One would initialize the stream, and then pass it to APIs that then use the base abstract Stream as input.

Microsoft's implementation chose to talk to the driver direct and have the buffering in the driver, where I chose to do buffering within software for better abstraction of the different problems between libraries.

closed 2021-07-11 by jcurl