jcurl/RJCP.DLL.SerialPortStream #125: Read takes 30ms?

WilliamForsdal, 2021-09-27:
Hey,
When testing this with our application (on Windows using net5) I get ~30ms time to read from the port. Is this expected with this library? Or have I configured it in a way that causes this? Could dotnet 5 be the issue?
These are my settings (I use the same when using the regular `System.IO.Ports.SerialPort` and I get <1ms read time):
```csharp
public static SerialPortStream CreateSerialPort(string portName)
{
    // default port settings
    const int      DefaultBaudRate     = 115200;
    const Parity   DefaultParity       = Parity.None;
    const int      DefaultDataBits     = 8;
    const StopBits DefaultStopBits     = StopBits.One;
    const bool     DefaultDtrEnable    = true;
    const int      DefaultWriteTimeout = 100;
    const int      DefaultReadTimeout  = 100;
    return new SerialPortStream(portName, DefaultBaudRate, DefaultDataBits, DefaultParity, DefaultStopBits)
    {
        WriteTimeout = DefaultWriteTimeout,
        ReadTimeout  = DefaultReadTimeout,
        DtrEnable    = DefaultDtrEnable
    };
}
```
Happy to provide more info if needed.

jcurl, 2021-09-27:
Hi, when you measure the 30ms, from when do you measure it?

The data is buffered (assuming you're using Windows) via a dedicated thread. If the data is already in the buffer, then it should be sub-millisecond.

The library is not designed for low latency.

So, the data probably hasn't arrived yet. When it does, it is read into the buffer and depends on Windows APIs. If your thread is already blocked because no data was present.

There a couple of places where the delays can occur.

1. Probably, the driver might return after a timeout, to make sure new data isn't arriving. This is probably configurable in the low level code. Look in CommOverlappedIo and around there on how the serial port is initialised.

2. There would also be a few context switches needed. The user thread is likely blocked, and the CommOverlappedIo signals when the data arrived.

Hope this helps.

WilliamForsdal, 2021-09-27:
Thanks for the quick reply!
Measuring with a StopWatch around the Read method:
```csharp
Stopwatch sw = Stopwatch.StartNew();
int bytesRead = Stream.Read(_readBuf, _readPtr, 64);
sw.Stop();
Console.WriteLine($"Read {bytesRead} after {sw.ElapsedMilliseconds}");
```
sw.ElapsedMilliseconds is 29-31ms every time. It's suspiciously close to the regular windows timeslice of 15ms times two, so perhaps it's due to some context switching like you say.
Was hoping not to get into the low level stuff for this, so I'll stick with the regular SerialPort for now. Thanks again.

closed 2021-09-27 by WilliamForsdal

jcurl, 2021-09-27:
You might still want to look at Timeouts in the CommOverlappedIo when reading data. You can enable tracing.

If you follow the wiki, it describes how to do this with the TraceSource.

If you use .NET Core, you can inject an ILogger. This might confirm that the delays are not due to the Serial port configuration.