jcurl/RJCP.DLL.SerialPortStream #131: DataReceived is not triggered

Rampftl, 2021-12-06:
Hi, 
i have a problem triggering the DataReceived Event. I installed virtual serial port tools (from HHD) and created a local bridge between COM2 and COM3. I created two small TestApps. One connects to COM2 and sends a string. The other connects to COM3 and receives data.

SerialTest:
`using System;
using RJCP.IO.Ports;

namespace SerialTest
{
    class Program
    {
        static void Main(string[] args)
        {
            SerialPortStream serialPortStream = new SerialPortStream("COM2", 115200, 8, Parity.None, StopBits.One);
            serialPortStream.Open();
            if (serialPortStream.IsOpen)
            {
                Console.WriteLine("Port is open");
                string sendString = "blablabla";
                serialPortStream.Write(sendString + 0x1A);
                Console.WriteLine("Data sent");
            }
            else
            {
                Console.WriteLine("Port is not open");
            }
        }
    }
}`

SerialTest2:
`using System;
using RJCP.IO.Ports;

namespace SerialTest2
{
    class Program
    {
        static void Main(string[] args)
        {
            string readBuf = null;
            SerialPortStream serialPortStream = new SerialPortStream("COM3", 115200, 8, Parity.None, StopBits.One);
            serialPortStream.DataReceived += delegate(object sender, SerialDataReceivedEventArgs eventArgs)
            {
                Console.WriteLine("sender = {0}, args = {1}", sender.ToString(), eventArgs.ToString());
            };
            serialPortStream.Open();
            if (serialPortStream.IsOpen)
            {
                Console.WriteLine("Port is open");
                
                
                while(string.IsNullOrEmpty(readBuf))
                {
                    // Do nothing
                }
                
            }
            else
            {
                Console.WriteLine("Port is not open");
            }
        }
    }
}`

I start SerialTest2 which has a delegate on the DataReceived event. Then I start SerialTest1 which sends the string (including the EOF). The delegate in SerialTest2 is not invoked.

Do I use the SerialPortStream the wrong way?

Best
Mic

jcurl, 2021-12-31:
Please test again using 2.4.0 and report your results. Thank you!

jcurl, 2022-02-08:
Closing as there is no feedback.

closed 2022-02-08 by jcurl