jcurl/RJCP.DLL.SerialPortStream #132: Unhandled exception. System.NullReferenceException

ilrom, 2021-12-31:
I use the library, an error occurs only on Linux machines
`Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
at RJCP.IO.Ports.Native.SerialBuffer.RJCP.IO.Ports.Native.ISerialBufferStreamData.get_BytesToRead()
at RJCP.IO.Ports.SerialPortStream.HandleEvent(Object state)
at System.Threading.QueueUserWorkItemCallback.<>c.<.cctor>b__6_0(QueueUserWorkItemCallback quwi)
at System.Threading.ExecutionContext.RunForThreadPoolUnsafe[TState](ExecutionContext executionContext, Action`1 callback, TState& state)`
Which leads to the failure of my program.
- This error does not occur often when the device is reconnected (physically).
- When it opens often, the port closes.

jcurl, 2021-12-31:
Thank you for the bug report. Can you tell me the version of the nserial, the library from .net, and the run time you are using?

ilrom, 2021-12-31:
Thanks for the quick response
<PackageReference Include="SerialPortStream" Version="2.3.1" />
 run time 3.1.17

jcurl, 2021-12-31:
I've just posted version 2.4.0 which has changes for handling exceptions in the callbacks. Perhaps this fixes your problem?

If the problem persists, you'll need to enable logging (e.g. console logging), and inject that logger into the `SerialPortStream` constructor.

ilrom, 2022-01-01:
Thanks, I will test and answer whether it helped me.

Muximize, 2022-02-04:
@ilrom Does 2.4.0 fix your issue?

jcurl, 2022-02-04:
I analysed the code again, and the problem should not occur with 2.4.0. The callback for the handling of events has now a try/catch block (which would cause .NET Core program to exit, instead of causing an exception).

I'll leave this ticket open for a few days, otherwise I'll close it as fixed in 2.4.0 which is now a NuGet package for a month or so.

closed 2022-02-08 by jcurl