jcurl/RJCP.DLL.SerialPortStream #134: Can't compile - missing source code

hallsbyra, 2022-04-25:
Where can I find these projects, referenced by SerialPortStream.csproj, but not included in this repo:

```
  <ItemGroup>
    <ProjectReference Include="..\..\bufferio\BufferIO\RJCP.IO.Buffer.csproj" />
    <ProjectReference Include="..\..\trace\Trace\RJCP.Diagnostics.Trace.csproj" />
  </ItemGroup>
```

hallsbyra, 2022-04-25:
Ok, just found'em! :)

* https://github.com/jcurl/RJCP.DLL.BufferIO
* https://github.com/jcurl/RJCP.DLL.Trace

closed 2022-04-25 by hallsbyra

jcurl, 2022-04-25:
The idea is to check out https://github.com/jcurl/RJCP.base and do a submodule init. Then you can be sure that the interfaces between the repositories are compatible. It also of course sets up the folder structure. Building can still be done in the repo directly with `dotnet build`.

You could also check out the `v2.x` branch which doesn't have these dependencies.