jcurl/RJCP.DLL.SerialPortStream #138: Version 3.x ?

StefH, 2022-12-20:
The NuGet for SerialPortStream describes:

> An independent implementation of System.IO.Ports.SerialPort and SerialStream for better reliability and maintainability. Version 3.x and later will be under the package name RJCP.SerialPortStream.

However, I cannot find RJCP.SerialPortStream at all?

jcurl, 2022-12-20:
For now, just use SerialPortStream, which is 2.x. The new version 3.x will have a bunch of new dependencies which haven't been released yet as I did refactoring. The functionality between the two are the same.

closed 2022-12-20 by jcurl

tyouard, 2024-04-07:
Hi Jason. Is this advice still valid? I have been using 3.x version but just noticed that 2.x release is listed on the repo main page.

jcurl, 2024-04-08:
I will continue with version 2.x for bugfixes only and it won't receive any major refactoring. Version 3.x is updated for newer versions of .NET Core.

It does look like I forgot to make the "release" for tags 3.0.0 and 3.0.1.