jcurl/RJCP.DLL.SerialPortStream #140: Incompatible with VS 2022

arthurmelow, 2023-02-14:
Hii,

I need to work with visual studio 2022 with .net framework 4.8, windows 11 and i must use the SerialPortStream dll. I can't compile the code because there are some errors about the compatibility with:

- netstandard2.0
- .net framework 4.8
- Visual Studio 2022 -> don't support .net framework 4.5 and net core downer than 2.1 https://learn.microsoft.com/es-es/visualstudio/releases/2022/compatibility 

Could you indicate if is this dll compatible with my setup ?? And, In case of yes, how can i do to compile and use this dll ??

Thanks and regards,

jcurl, 2023-02-14:
Hi, as per MS docs, you need also Visual Studio 2019 to provide .NET 4.0 SDK and other target SDKs. The project still supports older OSes. 

I compile everything fine on Win10 and Win11 from the command line with VS2022.

arthurmelow, 2023-02-15:
And, how do you resolve the problem with LogSource ?

jcurl, 2023-02-16:
Hello, if you check out the branch v2.x, then you won't have this dependency, and can compile.

If you wish to compile the branch master, which is a refactoring of 2.x, you will need to check out the RJCP.base repository, initialise all sub modules. Then you can enter the framework folder and do a "dot net build" from there. It references the LogSource which is a different repository of mine.

jcurl, 2023-02-26:
Hi closing as there is no feedback. If you have any questions, or need more details, please write back.

closed 2023-02-26 by jcurl