jcurl/RJCP.DLL.SerialPortStream #142: https://github.com/jcurl/SerialPortStream: Problem with RD-Server Port-Redirection

kunostoeckli, 2024-02-07:
Hi Jason, 

Thanks for the RJCP.DLL.SerialPortStream. 

We searched a library for Windows Serial-Ports, which uses Kernel32-File-Read-Functionality. 

On some Windows 2022 RD-Server, after a ReadExisting(), the connection got closed. 

(Technically: in RJCP.IO.Ports.Native.Windows.CommOverlappedIo, the OverlappedIoThread() got an error and then m_IsRunning = false; 
)

I think we found a solution: 
after  Kernel32.GetOverlappedResult(m_ComPortHandle, ref readOverlapped, out bytes, true);
we get the error 38 (WinError.ERROR_HANDLE_EOF) on the RD-Server. 

We think, that this EOF is a legal answer. We tried it in the Version 2.x branch as following: 
                        `    if ((win32Error != WinError.ERROR_OPERATION_ABORTED && win32Error != WinError.ERROR_HANDLE_EOF) || bytes > 0)  {`

What do you think: Is the additional `( .. && win32Error != WinError.ERROR_HANDLE_EOF)` correct?
If yes: Can you add this to your repository? 

Thanks ans best regards!

Kuno. 

[image]

jcurl, 2024-02-07:
Looking a bit at the code, it might be related to the character 0x1A arriving over the serial port. As I've never been able to reproduce this with the drivers that I have, could I ask you please to:
a) Let me know the serial port. Is this something I can perhaps buy from Amazon to test?
b) With your proposed changes, are you able to confirm all data arriving? Or would it drop the byte 0x1A?
c) Do you get an event SerialData.Eof from SerialPortStream.DataReceived?

The code change could be made, but then I wonder if an event would be missed, so this will need proper testing.

Regarding an update, I'm in the process of upgrading SerialPortStream 3.x to .NET Core 6.0 (dropping .NET Standard 2.1), so I'll get that work done first, then I hope to test for this change.

kunostoeckli, 2024-02-08:
Hi Jason, 

 

a)

We use a Zebra Barcode-Scanner (For example DS2208) which is configured as a serial-port device. 

In the attachment is the configuration with which we configure the scanners, so they behave as a serial-port-device. 

 

The scanner is attached to an PC with Windows 11. 

The Windows 11 PC connects to a Windows 2022 Remote-Desktop-Server with Port-Redirection. 

On the Windows 2022 server, the port is visible with the command “mode” in a command box. 

 

We receive the error after the first scan. 

This is returned correctly. But then "IsOpen" goes to False

 

 

b) 

It looks as if we get everything with the proposed change. 

 

c) 

No. 

We get only with the Kernel32. GetOverlappedResult() this WinError. (

 

Fix: 

Yes we’ve seen that the code is in version 3. But if we download version 3 Source-Code then not everything is included. That’s why we changed it in Version 2.x

We imagine this file stream in such a way that we are just at the end until something is scanned again. And from our point of view, this end is good.

 

Best regards

 

Kuno.

jcurl, 2024-02-08:
Thanks for the answer. I'll plan to do the update in both 2.x and 3.x

To check out the 3.x code, you must:
* git checkout RJCP.DLL.SerialPortStream serialportstream
* git checkout RJCP.DLL.BufferIO bufferio
* git checkout RJCP.DLL.Trace trace
* git checkout RJCP.DLL.DeviceMgr devicemgr

Change the RJCP.DLL to include your preferred URL (https, ssh, or whatever). Ensure they're all alongside each other. Then in the `serialportstream` folder, a `dotnet build` should be sufficient, and the https://github.com/jcurl/RJCP.DLL.SerialPortStream/blob/master/code/SerialPortStream.csproj finds the relative paths and builds. Everything on HEAD master is fine.

I might have to look into a workaround for the driver giving EOF, but not raising the event. I would have expected the event to be raised.

kunostoeckli, 2024-02-09:
Hi, 

 

Thanks a lot!

 

Best regards

 

Kuno.

jcurl, 2024-02-21:
Hello, I would appreciate if you would test the branch bugfix/dotnet-975 (based on v2.x). I've provided a fix and refactored slightly to keep code readable.

I will do manual testing on the weekend (I don't have a way to reproduce this issue), and will check for any regressions I might find. All my unit tests pass.

Then after this is tested, I'll backport the changes on to the 3.x branch and provide NuGet releases for all.

kunostoeckli, 2024-02-23:
Hi Jason, 

 

Thanks for your informations. 

 

I hope, I’ll find some time next week to test the branch. Then I’ll give you feedback. 

 

Best regards

 

Kuno.

jcurl, 2024-03-04:
Hi, I appreciate your busy. I'm nearly ready for a new release of all my packages, pending if you have feedback.

kunostoeckli, 2024-03-04:
Hi, 
Do you have https://www.buymeacoffee.com/ or something similar. 
Best regards
Kuno.

jcurl, 2024-03-04:
Thank you! I appreciate your gesture. One of my biggest hopes in Open Source are contributes from people to make the software better. Thank you for not only reporting the problem, but suggesting the solution.

jcurl, 2024-03-04:
Now released with version 2.4.2 and uploaded in NuGet.

closed 2024-03-04 by jcurl

kunostoeckli, 2024-03-04:
Thanks a lot!

kunostoeckli, 2024-03-07:
Hi, 

 

It works now good with the NuGet-Version 2.4.2. 

 

Thanks a lot!

 

Best regards

 

Kuno.