jcurl/RJCP.DLL.SerialPortStream #143: SerialThread: Died from Access is denied

JaredWF, 2024-03-02:
I've been trying to debug a flakiness issue that happens for a couple of my users. After some amount of time of continuous reading and writing, the `IsOpen` flag changes to false and I can no longer read or write on the port. I've traced it through to `SerialPortStream.cs`, and when the port closes itself, `m_NativeSerial.IsOpen` is true while `m_NativeSerial.IsRunning` is false. That led to finding the error `COM3: SerialThread: Died from Access is denied. (Exception from HRESULT: 0x80070005 (E_ACCESSDENIED))`. I'm not able to replicate this exact scenario on my machine, but I did notice that the same error throws if I unplug my device. It's just weird that the error happens for other users while the device is still connected. Any idea why this happens and how to properly recover from this state? Close and reopen the port?

jcurl, 2024-03-04:
Hi, thank you for the report. I appreciate you've seen the logs and were able to identify the root cause that the driver is giving a specific error (which is why it leads to stop running the device, as there is no guarantee that further win32 API calls would work). Have you checked the Windows Event Log if there reports raised by the USB device? e.g. removal, insertion? My experience is that USB devices are sensitive to cable quality, external USB hubs. My tests with PL2303 or FTDI haven't shown this issue.

JaredWF, 2024-03-17:
Sorry, I missed this reply! That could be it, but nobody has mentioned any disconnect noises from Windows when this issue happens. For checking events raised by the USB device, would that be in event viewer in the Windows Logs->System section?

jcurl, 2024-03-18:
Hello, I didn't find much on how to do this via Event Viewer, although a few posts suggest this is possible. There appears to be a program by NirSoft that can log USB events also. Else WireShark takes USB logs, but is probably too much data for what you need here.

jcurl, 2024-03-20:
https://www.nirsoft.net/utils/usb_log_view.html seems to work.

e.g.
[image]

JaredWF, 2024-03-21:
That's perfect, thanks so much! You can probably close this issue for now, it'll take time to gather more info from our users.

closed 2024-03-23 by jcurl