jcurl/RJCP.DLL.SerialPortStream #144: In Ubuntu 18.04.4 LTS, BytesToRead is 0

ccvvplayer, 2024-03-26:
Dear jcurl, my program framework is netcoreapp3.1, running in windows can normally read PLC data, but in linux will appear BytesToRead 0, can not read the data situation. In fact, the PLC device has a response in the serial port monitoring.
string portName = "/dev/ttysWK0"; 
            SerialPortStream serialPort = new SerialPortStream(portName);

            serialPort.BaudRate = 9600;  
            serialPort.Parity = RJCP.IO.Ports.Parity.Odd;  
            serialPort.DataBits = 8;  
            serialPort.StopBits = RJCP.IO.Ports.StopBits.One;  
            serialPort.DtrEnable = true;
            serialPort.RtsEnable = true;
                    serialPort.Open();  
                var send = new byte[] { 1, 3,1,194, 0, 1 }.AppendCrc();
                    serialPort.Write(send,0,send.Length);
                serialPort.DataReceived += DataReceivedHandler;
                Console.ReadKey();
private static void DataReceivedHandler(object sender, RJCP.IO.Ports.SerialDataReceivedEventArgs e)
        {
            SerialPortStream port = (SerialPortStream)sender;
            int bytesToRead = port.BytesToRead;
            byte[] readBuffer = new byte[bytesToRead];
            port.Read(readBuffer, 0, bytesToRead);
            Console.WriteLine($"{bytesToRead}");
        }
Here's another way to read it.

              byte[] received = null;
                
                if (SpinWait.SpinUntil(() => serialPort.BytesToRead >= 1, 1000))
                {
                 Console.WriteLine($"receives data：{serialPort.BytesToRead}");
                 received = new byte[serialPort.BytesToRead];
                 serialPort.Read(received, 0, received.Length);
                  Console.WriteLine($"receives data：{serialPort.BytesToRead}");
                }
                else
               {
                  Console.WriteLine($"no receives:{0}",serialPort.BytesToRead);
               }

jcurl, 2024-03-26:
Hi! Could you please provide logs from the SerialPortStream? What version of SerialPortStream are you using? Also, for this to properly work on Linux, you will need my native library libnserial. Which version of this are you using?

For me to test, are you able to provide the smallest project needed to reproduce? Then I can test on my machine.

ccvvplayer, 2024-03-27:
Hello, jcurl. My Version information is RJCP.SerialPortStream, Version=2.4.2.0. I have found the reason why I could not read the data in PLC today, I changed the PLC parity check to None, and BytesToRead received the data. It's not clear what caused it.

ccvvplayer, 2024-03-27:
> 你好！您能否提供 SerialPortStream 的日志？您使用的是哪个版本的 SerialPortStream？此外，为了在 Linux 上正常工作，您将需要我的本机库 libnserial。您使用的是哪个版本？
> 
> 为了让我测试，您是否能够提供复制所需的最小项目？然后，我就可以在我的机器上进行测试了。

Hello, jcurl. My Version information is RJCP.SerialPortStream, Version=2.4.2.0. I have found the reason why I could not read the data in PLC today, I changed the PLC parity check to None, and BytesToRead received the data. It's not clear what caused it.

ccvvplayer, 2024-03-27:
> Hi! Could you please provide logs from the SerialPortStream? What version of SerialPortStream are you using? Also, for this to properly work on Linux, you will need my native library libnserial. [...]

Hello, jcurl. My Version information is RJCP.SerialPortStream, Version=2.4.2.0. I have found the reason why I could not read the data in PLC today, I changed the PLC parity check to None, and BytesToRead received the data. It's not clear what caused it.

ccvvplayer, 2024-03-27:
> Hi! Could you please provide logs from the SerialPortStream? What version of SerialPortStream are you using? Also, for this to properly work on Linux, you will need my native library libnserial. [...]
I don't know if you understand my meaning, the problem I encountered is that on windows, I can read the data of device protocol 19200, o, 8 and 1, but on linux, BytesToRead is 0. I confirm that the parameter configuration before the serial port is opened is consistent with the PLC device communication protocol. Today, I tried to change the communication protocol of PLC equipment to 19200, n, 8, 1 to read it. My Linux architecture is aarch64, it is not clear whether it is a system issue or something else that is causing it. Thank you, jcurl.

jcurl, 2024-03-27:
I understand your meaning. But this is not compilable working code that I can use for the next step of debugging or confirming the problem. Logs when the problem occur have also not been provided, so I cannot guess as to what the root cause may be.

If it is important, please provide a minimal code example that compiles using netcoreapp3.1, so I can try to reproduce the problem. If I can, I will be able to research it and understand what's going wrong. If I can't reproduce the problem, then this indicates another issue, and I can advise on the next steps.

ccvvplayer, 2024-03-28:
> I understand your meaning. But this is not compilable working code that I can use for the next step of debugging or confirming the problem. Logs when the problem occur have also not been provided, [...]

Thank you, jcurl. But I can't provide.NET Core 3.1 compiled code, only C# code written using VS2022 tool. When the problem occurred, the program did not report any error or prompt, but the BytesToRead of SerialPortStream could not read the data, but my serial monitoring tool could see the data replied by the PLC. Currently the program can only read devices without parity.

jcurl, 2024-03-28:
Excellent, please provide the smallest example, of `.cs` and `.csproj` file that shows the problem, as you created using VS2022. I will run it using my tools on Ubuntu and Windows and see where the problem could lie.

ccvvplayer, 2024-03-29:
[test.zip](https://github.com/jcurl/RJCP.DLL.SerialPortStream/files/14797844/test.zip)

> Excellent, please provide the smallest example, of `.cs` and `.csproj` file that shows the problem, as you created using VS2022. I will run it using my tools on Ubuntu and Windows and see where the problem could lie.

jcurl, here are the cs and csproj files of the example code, but it is just the usual example code, the only problem is that it will work on Windows devices with parity, but not on linux.

jcurl, 2024-03-29:
Hello @ccvvplayer , I'm terribly sorry, can you try to upload again? For some reason, GitHub is saying "Not Found" when I click on https://github.com/jcurl/RJCP.DLL.SerialPortStream/files/14797844/test.zip

ccvvplayer, 2024-03-30:
> Excellent, please provide the smallest example, of `.cs` and `.csproj` file that shows the problem, as you created using VS2022. I will run it using my tools on Ubuntu and Windows and see where the problem could lie.

Ok, no problem
[test.zip](https://github.com/jcurl/RJCP.DLL.SerialPortStream/files/14811030/test.zip)

jcurl, 2024-04-07:
Dear @ccvvplayer, I've taken your test code, updated it [20240407_073bd2f.zip](https://github.com/jcurl/RJCP.DLL.SerialPortStream/files/14898014/20240407_073bd2f.zip), and tested on my machine running Ubuntu 22.04 with netcoreapp3.1 on my machine.

The test code works, such that sending 9600,8,O,1 works, and is received on another serial port.

I set up the following:
* On one USB port, a PL2303TA chip, connect the TX pin to the RX pin of
* another USB port, a PL2303TA chip

So that when I run the command, it sends 4 bytes (and waits to receive)
```sh
test1 /dev/ttyUSB1
```

Then I run the same command on another terminal that sends the bytes
```sh
test1 /dev/ttyUSB0
```

I get:
```
Testing on /dev/ttyUSB1
Opening /dev/ttyUSB1,8,O,1
Writing to /dev/ttyUSB1
Waiting for response on /dev/ttyUSB1
BytesToRead: 4
Read: 4
```

This indicates that the problem may be your serial port chipset, or the Linux kernel driver. I can suggest using a different chipset, or trying different drivers on Linux (or a newer Linux kernel version).

As there are a lot of reports suggesting that 8 databits with odd parity might not work, I also connected it to my oscilloscope, and confirmed that it did properly send the bytes on the wire. The "0" is at 0V, the "1" is at 3.3V. Each byte sequence is 11-bits as expected (1 start, 8 data bits, 1 parity, 1 stop bit).

```
01010101011
S5   5   OE

00101010111
SA   A   OE

00110100001
S6   1   OE

00101100111
SA   9   OE
```

This means there is nothing that I can do in software.

closed 2024-04-14 by jcurl

ccvvplayer, 2024-04-18:
Sorry, icurl. I was busy these days and didn't reply to the messages on git in time. I saw your response. The problem is not that I read it before printing, but that serialPort.BytesToRead has no bytes at all. At present, I do not know where the problem is, but still according to the way I started to talk about can read the data, only the PLC communication protocol does not set parity can read the data.

> 亲爱的，我已经接受了您的测试代码，[对其进行了20240407_073bd2f.zip](https://github.com/jcurl/RJCP.DLL.SerialPortStream/files/14971214/20240407_073bd2f.zip)更新，并在运行 Ubuntu 22.04 的机器上进行了测试，并在我的机器上安装了 netcoreapp3.1。
> 
> 测试代码有效，因此发送 9600,8，O，1 有效，并在另一个串行端口上接收。
> 
> ## 测试顺序
> 我设置了以下内容：
> 
> * 在一个 USB 端口 （） 上，一个 PL2303TA 芯片，将 TX 引脚连接到`/dev/ttyUSB0`
> * 另一个 USB 端口 （），一个 PL2303TA 芯片`/dev/ttyUSB1`
> 
> 因此，当我运行该命令时，它会发送 4 个字节（并等待接收）
> 
> ```shell
> test1 /dev/ttyUSB1
> ```
> 
> 然后，我在另一个发送字节的终端上运行相同的命令
> 
> ```shell
> test1 /dev/ttyUSB0
> ```
> 
> 我得到：
> 
> ```
> Testing on /dev/ttyUSB1
> Opening /dev/ttyUSB1,8,O,1
> Writing to /dev/ttyUSB1
> Waiting for response on /dev/ttyUSB1
> BytesToRead: 4
> Read: 4
> ```
> 
> 这表明问题可能出在串行端口芯片组或 Linux 内核驱动程序上。我可以建议使用不同的芯片组，或在 Linux（或更新的 Linux 内核版本）上尝试不同的驱动程序。
> 
> 由于有很多报告表明奇偶校验的 8 个数据位可能不起作用，因此我还将其连接到我的示波器，并确认它确实正确地发送了网络上的字节。“0”为0V，“1”为3.3V。每个字节序列是 11 位（1 个启动位、8 个数据位、1 个奇偶校验、1 个停止位）。
> 
> ```
> 01010101011
> S5   5   OE
> 
> 00101010111
> SA   A   OE
> 
> 00110100001
> S6   1   OE
> 
> 00101100111
> SA   9   OE
> ```
> 
> 这意味着我在软件中无能为力。
> 
> ## 关于测试代码的一些说明
> 附加的 ZIP 文件实际上是一个 GIT 存储库。您将看到我所做的更改。
> 
> * 我只是等待一个字节，因为描述说没有收到任何字节
> * 您在读取字节_后_打印，这将返回 0（没有要读取的字节，因为您已经读取了它们）。在读取之前打印它实际上显示了正确的字节数，即 4。`serialPort.BytesToRead`
> * 我不得不删除硬件流量控制，因为我的串行端口没有硬件流量控制引脚，只有 TXD、RXD、GND。`DtrEnable``RtsEnable`

> Dear @ccvvplayer, I've taken your test code, updated it [20240407_073bd2f.zip](https://github.com/jcurl/RJCP.DLL.SerialPortStream/files/14971214/20240407_073bd2f.zip), and tested on my machine running Ubuntu 22.04 with netcoreapp3.1 on my machine.
> 
> The test code works, such that sending 9600,8,O,1 works, and is received on another serial port.
> 
> ## Test Sequence
> I set up the following:
> 
> * On one USB port (`/dev/ttyUSB0`), a PL2303TA chip, connect the TX pin to the RX pin of
> * another USB port (`/dev/ttyUSB1`), a PL2303TA chip
> 
> So that when I run the command, it sends 4 bytes (and waits to receive)
> 
> ```shell
> test1 /dev/ttyUSB1
> ```
> 
> Then I run the same command on another terminal that sends the bytes
> 
> ```shell
> test1 /dev/ttyUSB0
> ```
> 
> I get:
> 
> ```
> Testing on /dev/ttyUSB1
> Opening /dev/ttyUSB1,8,O,1
> Writing to /dev/ttyUSB1
> Waiting for response on /dev/ttyUSB1
> BytesToRead: 4
> Read: 4
> ```
> 
> This indicates that the problem may be your serial port chipset, or the Linux kernel driver. I can suggest using a different chipset, or trying different drivers on Linux (or a newer Linux kernel version).
> 
> As there are a lot of reports suggesting that 8 databits with odd parity might not work, I also connected it to my oscilloscope, and confirmed that it did properly send the bytes on the wire. The "0" is at 0V, the "1" is at 3.3V. Each byte sequence is 11-bits as expected (1 start, 8 data bits, 1 parity, 1 stop bit).
> 
> ```
> 01010101011
> S5   5   OE
> 
> 00101010111
> SA   A   OE
> 
> 00110100001
> S6   1   OE
> 
> 00101100111
> SA   9   OE
> ```
> 
> This means there is nothing that I can do in software.
> 
> ## Some notes about your test code
> The attached ZIP file is actually a GIT repository. You'll see the changes I made.
> 
> * I just wait for one byte, because the description said no bytes were received
> * You printed `serialPort.BytesToRead` _after_ reading the bytes, which will return 0 (there are no bytes to read, because you've read them). Printing this before reading actually shows the correct number of bytes, which is 4.
> * I had to remove hardware flow control `DtrEnable` and `RtsEnable` because my serial ports don't have hardware flow control pins, only TXD, RXD, GND.

Sorry, icurl. I was busy these days and didn't reply to the messages on git in time. I saw your response. The problem is not that I read it before printing, but that serialPort.BytesToRead has no bytes at all. At present, I do not know where the problem is, but still according to the way I started to talk about can read the data, only the PLC communication protocol does not set parity can read the data.

jcurl, 2024-04-18:
Dear @ccvvplayer, my test shows that `BytesToRead` is working and reading data, and it is based on the test code you provided, which is why I closed the issue. You also indicated that it works if you change the parity - so I tested the send and receive and made sure that the electrical signals are correct. Many posts on the Internet suggest that Odd parity with 8 data bits may not work on Linux.

my conclusion is:
* on my test system, Ubuntu 20.04, with a PL2303TA it works
* Using an oscilloscope I confirmed the protocol on the wire works
* The problem may be with Ubuntu 18.04 or the driver you use, in which case I don’t have a solution in my software.

I would suggest that you try the example code I provided, and see if you can reproduce my results. If not, that confirms a system problem. If it does work, maybe the problem is somewhere else.

ccvvplayer, 2024-05-08:
> Dear @ccvvplayer, my test shows that is working and reading data, and it is based on the test code you provided, which is why I closed the issue. You also indicated that it works if you change the parity - so I tested the send and receive and made sure that the electrical signals are correct. Many posts on the Internet suggest that Odd parity with 8 data bits may not work on Linux.`BytesToRead`
> 
> my conclusion is:
> 
> * on my test system, Ubuntu 20.04, with a PL2303TA it works
> * Using an oscilloscope I confirmed the protocol on the wire works
> * The problem may be with Ubuntu 18.04 or the driver you use, in which case I don’t have a solution in my software.
> 
> I would suggest that you try the example code I provided, and see if you can reproduce my results. If not, that confirms a system problem. If it does work, maybe the problem is somewhere else.

Ok, dear jcurl. Although the problem is closed, it does not prevent it from being visited. Hope you can meet other brothers who have the same problem as me on it. Let's discuss how to solve it. At present, I am going to switch to another hardware device and install another version of Ubuntu, and try to find out what the cause is. Because of my limited programming skills, I cannot contribute to the SerialPortStream library with you. Only here I wish you a smooth work and a happy life. jcurl.

jcurl, 2024-05-08:
I understand your situation. You'd have to debug it and provide more information on your set up to identify the fault. If the issue is not reproducible on my side, there's not much more I can do than give you tips on what else you can do.

You could attempt to rebuild "libnserial.so" with debugging enabled. For this see [CMakeLists.txt](https://github.com/jcurl/RJCP.DLL.SerialPortStream/blob/master/dll/serialunix/libnserial/CMakeLists.txt) where you need to enable the option `NSLOG_ENABLED`. Then when you run the test, it will write a log to `/tmp/nserial.log` as given at [log.c](https://github.com/jcurl/RJCP.DLL.SerialPortStream/blob/master/dll/serialunix/libnserial/log.c#L54).

If you see bytes arriving there, but not in the .NET implementation, this will give hints on the next debugging step. Good luck!

ccvvplayer, 2024-05-09:
> I understand your situation. You'd have to debug it and provide more information on your set up to identify the fault. If the issue is not reproducible on my side, there's not much more I can do than give you tips on what else you can do.
> 
> You could attempt to rebuild "libnserial.so" with debugging enabled. For this see [CMakeLists.txt](https://github.com/jcurl/RJCP.DLL.SerialPortStream/blob/master/dll/serialunix/libnserial/CMakeLists.txt) where you need to enable the option . Then when you run the test, it will write a log to as given at [log.c](https://github.com/jcurl/RJCP.DLL.SerialPortStream/blob/master/dll/serialunix/libnserial/log.c#L54).`NSLOG_ENABLED``/tmp/nserial.log`
> 
> If you see bytes arriving there, but not in the .NET implementation, this will give hints on the next debugging step. Good luck!

Thank you for your prompt, but at present my technology is not able to understand the operation of your prompt. But I'm going to try.