jcurl/RJCP.DLL.SerialPortStream #149: Bytes to Read is always 0

Michael-Biernacki, 2024-08-06:
I'm working with an Arduino and attempting to read it in through a COM Port.  I have it transmitting a bytes array constantly through the COM port.  I have given the same baud rate and the correct COM port to the SerialPortStream() constructor.  However, it is showing that there are 0 bytes to read.

I'm working with .NET standard 2.1 and Windows 11.

jcurl, 2024-08-22:
Do you have any logs being captured? As you're using .NET Core, you'll have to programmatically enable the logs. You'll need to read the [README.md](https://github.com/jcurl/RJCP.DLL.SerialPortStream?tab=readme-ov-file#613-net-core) that tells you how to do this.

You can also try using tools from SysInternals to see the results of the driver APIs.

Also note, that .NET Standard 2.1 isn't the version of the .NET Core you've installed (it's an API revision implemented by all versions of .NET Core 2.1, 3.0, 3.1, 5.0, 6.0, 7.0, 8.0, 9.0).

jcurl, 2024-08-27:
In another comment, note there is no fix as the root cause is unknown. My analysis then was a badly behaving driver that reported a fatal error. 

You will need to collect at a minimum .NET logs (see previous comment). You should also use SysInternal tools and look at the device events (e.g. you f there were any unplug events observed or other behaviour).