jcurl/RJCP.DLL.SerialPortStream #150: Do I still need to compile libnserial.so if using .net framework on linux?

rathbone01, 2024-08-27:
In the documentation it appears that you need to compile libnserial.so if using linux, is that only for mono, or also for a .net framework build with a linux target?

If it is still needed is there documentation somewhere on how to do that?

Thanks!

jcurl, 2024-08-28:
You will need to compile `libnserial.so` for .NET Mono and .NET Core. If you're using Debian / Ubuntu, you can install the debian files attached with the releases. If you'd like to build it yourself, see the documentation at [BUILD.md](https://github.com/jcurl/RJCP.DLL.SerialPortStream/blob/master/dll/serialunix/BUILD.md). There are other `README` files providing additional information.

It uses `cmake`, so follow the usual rules for building for your platform.

rathbone01, 2024-08-29:
Alright, thank you for your help!

closed 2024-08-29 by rathbone01