jcurl/RJCP.DLL.SerialPortStream #151: ReadAsync() - Difference between v2.4.2 and v3.0.1

AndreaCashItaly, 2024-09-16:
Good morning,
I just upgraded my solution with the newer version v3.0.1 (my software was still working with the previous v.2.4.2).

I get an error in the ReadAsync function.. is not working anymore.

I have an async Task loop that check everytime for bytes received. As soon upgraded, the ReadAsync function got into exception ("object not set to an instance of an object") when the receive buffer is empty. If some data is receive, I have no exception, but the int result is always 0 (zero bytes read).

What is changed with the new versione v3.0.1?

Thank you in advance

```
private async System.Threading.Tasks.Task leggi_risposta_ReadAsync()
{
        do
        {
            if (!this._port.IsOpen)
                break;

            // async bytes read into tmp buffer with fixed size (256 bytes)
            byte[] buffer_tmp = new byte[ITL_Utility_SPO.BUFFER_LENGTH];
            int actualLength = 0;
            try
            {
                actualLength = await this._port.ReadAsync(buffer_tmp, 0, buffer_tmp.Length).ConfigureAwait(false);
            }
            catch (Exception ex_inner1)
            {
                this._log_utility.LogERR(DateTime.Now, $"Errore leggi_risposta_ReadAsync(INNER#1) => {ex_inner1}");
            }

            for (int i = 0; i < actualLength; i++)
            {
                // do some stuff with received bytes...
            }
        } while (this._port.IsOpen);
}
```

jcurl, 2024-09-16:
Hello, and thanks for your report. I would be interested to solve it, but I need some information first.
* What .NET version are you using? The NuGet package offers four different versions (Net40, Net462, Net60, Net80)
* Can you please provide a precise stack trace of the exception?
* Would you be helpful in providing the minimum test program to reproduce the problem? This can help me debug it faster.

I keep both branches in sync with bugfixes and implementations, so there _should_ be no difference, but obviously you've observed a difference.

In principal, they should have the same functionality, but there v3.0.1 has been refactored and split into multiple libraries.

AndreaCashItaly, 2024-09-17:
Sorry to bother you here in this section, but I've been trying to update to v3.0.1 for a while now and wanted to try to resolve this error situation.

This morning, while I was getting the information you requested, I noticed a difference in my software.

I'll try to explain why the 'ReadAsync' function doesn't work for me in a specific circumstance: in the main UI, I call a function called 'Connection()' (located in a dll that manages devices). Inside this 'Connection()' function, the async loop task 'read_response_ReadAsync()' is called/started.

**OK** => if the 'Connection()' function from the main UI is called _through a BackgroundWorker_ (so asynchronously), everything works fine

**KO** => if the 'Connection()' function from the main UI is called _in the main thread_ (for example: 'Form_Load()'), the 'ReadAsync' function throws me an exception or, after the set timeout, returns me always 0 bytes received

All in all, it's not a big problem, maybe you can just ignore this **_"issue"_**.
I apologize again.

(in the whole software, I made so many calls to the 'Connection()' function. All of them are done through BackgroundWorkers, except one! Unfortunately, that one is the easiest to test.. so I ran into the problem right away, without going further into the other sections of the software)

jcurl, 2024-09-17:
Hello, not a problem! It's useful to try and find out the root cause, the update should not lead to such a big effort. So this concerns me.

From what you describe, have you tried to configure the `.ConfigureAwait(false)` from the invocation in the UI thread? Perhaps it is entering some errors through the way tasks work.

I did notice this some time back (all my testing is not done from a UI, so that could explain why I have not observed the issue), and the latest commit has the fixes. I'd have to rebuild to see if this makes the change (it's not complicated, but need to check out multiple repositories).

jcurl, 2025-04-25:
I've provided the latest release 3.0.2 that uses `ConfigureAwait(false)` internally.

closed 2025-04-25 by jcurl